Entware/Entware #381: Package Request: Package Request:

tomyang001, 2019-12-15:
Package request: [](https://github.com/pymumu/smartdns)

SmartDNS is a local DNS server. SmartDNS accepts DNS query requests from local clients, obtains DNS query results from multiple upstream DNS servers, and returns the fastest access results to clients.
Avoiding DNS pollution and improving network access speed, supports high-performance ad filtering.
Unlike dnsmasq's all-servers, smartdns returns the fastest access resolution.

ryzhovau, 2019-12-16:
Whats the killer feature of SmartDNS? We've got 5 or 6 DNS servers in repo, why we should add another one?

tomyang001, 2019-12-16:
It can test results from multiple upstream DNS server and choose the fastest one (lowest delay). Supports the latest TLS, HTTPS protocol and provides secure DNS query capabilities. DNS anti-poison mechanism, and a variety of mechanisms to avoid DNS pollution. These features are quite important for users in China for the terrible network environment.

closed 2020-01-31 by The-BB