Entware/Entware #384: Package request: bash-completion

bam80, 2019-12-21:
Package request
---
bash-completion: programmable completion functions for bash

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my {platform} and get a feedback about it.

---
The `bash-completion` was [deleted](https://github.com/Entware/entware.github.io/blob/master/_posts/2015-04-01-Changelog.md) in 2015 for unknown reasons.
Could we restore it?

ryzhovau, 2020-08-04:
It was proven as broken as long as OpenWrt (upstream) stripped this feature to make bash lighter/faster. Won't add it again, sorry.

closed 2020-08-04 by ryzhovau

Rotozaza, 2020-12-21:
Yes for https://github.com/scop/bash-completion

I run Entware on Qnap QTS using Entware Repo. Is OpenWrt in this setup involved?