Entware/Entware #385: Package update request: qbittorrent 4.2.1 and deluge 2.0.3

ted423, 2019-12-22:
I use qnap 453bmini . It seems qbittorrent  4.2.1's webUI is amazing.
like label and  Add ability to rename torrent files from the WebUI

papampi, 2020-01-07:
+1 Deluge 2.0.3

**Deluge 2.0 release notes**
**What’s new:**

Migrated to Python 3 with minimal support retained for Python 2.7.
Shiny new logo.
Multi-user support.
Performance updates to handle thousands of torrents with faster loading times.
A New Console UI which emulates GTK/Web UIs.
GTK UI migrated to GTK3 with UI improvements and additions.
Magnet pre-fetching to allow file selection when adding torrent.
Fully support libtorrent 1.2 release.
Language switching support.
Improved documentation hosted on ReadTheDocs.
AutoAdd plugin replaces built-in functionality.
Web UI now daemonizes by default so service scripts will require -d option.

ryzhovau, 2020-02-04:
Deluge 2.0.3 is there. 
Qbittorrent stays on 4.1.9.1, the newer version semms unstable.

closed 2020-02-04 by ryzhovau