Entware/Entware #387: coreutils has no files

bam80, 2019-12-30:
`coreutils` has no installed files, and the package size confirms it ("Size: 772"):
```sh
# opkg install coreutils
Installing coreutils (8.30-2) to root...
Downloading http://bin.entware.net/aarch64-k3.10/coreutils_8.30-2_aarch64-3.10.ipk
Configuring coreutils.

# opkg files coreutils
Package coreutils (8.30-2) is installed on root and has the following files:

# opkg info coreutils
Package: coreutils
Version: 8.30-2
Depends: libc, libssp, librt, libpthread
Status: install user installed
Section: utils
Architecture: aarch64-3.10
Size: 772
Filename: coreutils_8.30-2_aarch64-3.10.ipk
Description: Full versions of standard GNU utilities. If an equivalent Busybox applet is
 available, you should consider compiling that instead as Busybox applets are
 usually smaller, at the expense of reduced functionality.
Installed-Time: 1577725548

ryzhovau, 2019-12-30:
It's just a meta package. Please, use `opkg install coreutils-<applet>` to install necessary part of coreutils.

closed 2019-12-30 by ryzhovau

bam80, 2019-12-30:
Maybe point it in the Description? Right now, it seems a little confusing:
```
Description: Full versions of standard GNU utilities...