Entware/Entware #390: Tomato installation

moarf56, 2020-01-01:
Hi,

I try to use Entware on my FreshTomato Firmware 2019.3 K26ARM USB AIO-64K on a Asus RT-N18.
I follow the wiki page and I haven't any problem during the installation. Execpt at the end.

When the installtion is finish I can install or use OPKG whithout any issue, but when I make the verifycation procedure whith the reboot, the OPKG isn't stay : 
 
`root@unknown:/tmp/home/root# opkg list
-sh: opkg: not found
`

I don't have any error message during the installtion and I can install Lynx : 
`root@unknown:/tmp/home/root# opkg install lynx
Installing lynx (2.8.9rel.1-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/lynx_2.8.9rel.1-1_armv7-2.6.ipk
Installing libncursesw (6.1-5) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libncursesw_6.1-5_armv7-2.6.ipk
Installing zlib (1.2.11-3) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/zlib_1.2.11-3_armv7-2.6.ipk
Installing libopenssl (1.1.1d-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libopenssl_1.1.1d-1_armv7-2.6.ipk
Installing ca-bundle (20190110-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/ca-bundle_20190110-1_all.ipk
Configuring libncursesw.
Configuring zlib.
Configuring libopenssl.
Configuring ca-bundle.
Configuring lynx.
root@unknown:/tmp/home/root# lynx google.Fr
Looking up  'google.Fr' first
`

I try with a fresh freshtomato install but same thing. The df command return : 
`root@unknown:/tmp/home/root# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                23040     23040         0 100% /
devtmpfs                127808         0    127808   0% /dev
tmpfs                   127808       176    127632   0% /tmp
devfs                   127808         0    127808   0% /dev
/dev/sda1             15384960    118608  14484836   1% /tmp/mnt/ENTWARE
`
The /opt isn't mount so I try : 
`root@unknown:/tmp/home/root# mount /opt
mount: mounting LABEL=ENTWARE on /opt failed: No such file or directory`

What can I try to keep the installation fonctional after reboot ?

Thx :d

zyxmon, 2020-01-01:
Do not install to tmpfs. We do not provide support for any firmwares. It is better to ask at freshtomato forum!

closed 2020-01-01 by zyxmon

moarf56, 2020-01-01:
> 
> 
> Do not install to tmpfs. We do not provide support for any firmwares. It is better to ask at freshtomato forum!

Okay thx