Entware/Entware #391: jq not working on Asuswrt-Merlin 384.14 (RT-AC88U)

Xentrk, 2020-01-02:
I have a error with the jq package:
 
`# jq -r '.prefixes | .[].ip_prefix' < ip-ranges.json
jq: error while loading shared libraries: libonig.so: cannot open shared object file: No such file or directory
# jq
jq: error while loading shared libraries: libonig.so: cannot open shared object file: No such file or directory
# echo $PATH
/opt/bin:/opt/sbin:/bin:/usr/bin:/sbin:/usr/sbin:/home/wizard:/mmc/sbin:/mmc/bin:/mmc/usr/sbin:/mmc/usr/bin:/opt/sbin:/opt/bin:/opt/usr/sbin:/opt/usr/bin
# find / -name libonig.so

#`

ryzhovau, 2020-01-02:
```
opkg install oniguruma
```
Looks like `opkg` fails to detect new `jq` dependency. This error can't be reproduced on fresh install, so I'm closing this issue.

Xentrk, 2020-01-06:
Thank you @ryzhovau 

`opkg install oniguruma
Package oniguruma (6.9.3-1) installed in root is up to date.
# jq
jq: error while loading shared libraries: libonig.so: cannot open shared object file: No such file or directory
`

Xentrk, 2020-02-02:
I am now on 384.15 beta 1 release and applied the entware updates that were pushed yesterday. I still have the jq error message:
`jq: error while loading shared libraries: libonig.so: cannot open shared object file: No such file or directory`
From /jffs/scripts directory, I used the command below to see if libonig.so was installed:
`find / -iname libonig.so*`
The command returned:
`/tmp/mnt/entware/entware/lib/libonig.so.5`
I then navigated to the directory and issued the command below to copy libonig.so.5 to libonig.so:
`cp libonig.so.5 libonig.so`

jq now works!

ryzhovau, 2020-02-27:
Can't reproduce on fresh install:
```
~ # opkg install jq
Installing jq (1.6-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/jq_1.6-1_mipsel-3.4.ipk
Installing oniguruma (6.9.3-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/oniguruma_6.9.3-1_mipsel-3.4.ipk
Configuring oniguruma.
Configuring jq.
~ # ldd /opt/bin/jq
        linux-vdso.so.1 (0x771d0000)
        libm.so.6 => /opt/lib/libm.so.6 (0x77098000)
        libonig.so.5 => /opt/lib/libonig.so.5 (0x77005000)
        libgcc_s.so.1 => /opt/lib/libgcc_s.so.1 (0x76fdf000)
        libc.so.6 => /opt/lib/libc.so.6 (0x76e52000)
        /opt/lib/ld.so.1 (0x7719e000)
```
Please, reinstall `jq` and `oniguruma` and try again.

jackyaz, 2020-03-03:
@Xentrk you could use something like
```
opkg update && opkg remove --autoremove jq && opkg install jq
```

Xentrk, 2020-03-05:
> 
> 
> @Xentrk you could use something like
> 
> ```
> opkg update && opkg remove --autoremove jq && opkg install jq
> ```
Thank you Mr. Yaz,

The command solved the problem.  Thank you @jackyaz and @ryzhovau 
`opkg update && opkg remove --autoremove jq && opkg install jq
Downloading http://bin.entware.net/armv7sf-k2.6/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
Removing package jq from root...
oniguruma was autoinstalled and is now orphaned, removing.
Removing package oniguruma from root...
Installing jq (1.6-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/jq_1.6-1_armv7-2.6.ipk
Installing oniguruma (6.9.3-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/oniguruma_6.9.3-1_armv7-2.6.ipk
Configuring oniguruma.
Configuring jq.
wizard@RT-AC88U-8248:/jffs/scripts# jq
jq - commandline JSON processor [version 1.6]

Usage:  jq [options] <jq filter> [file...]
        jq [options] --args <jq filter> [strings...]
        jq [options] --jsonargs <jq filter> [JSON_TEXTS...]

jq is a tool for processing JSON inputs, applying the given filter to
its JSON text inputs and producing the filter's results as JSON on
standard output.

The simplest filter is ., which copies jq's input to its output
unmodified (except for formatting, but note that IEEE754 is used
for number representation internally, with all that that implies).

For more advanced filters see the jq(1) manpage ("man jq")
and/or https://stedolan.github.io/jq

Example:

        $ echo '{"foo": 0}' | jq .
        {
                "foo": 0
        }

For a listing of options, use jq --help.
`

closed 2020-03-05 by Xentrk