Entware/Entware #392: libcurl: curl-config missing

oke24, 2020-01-02:
On my Synology DS218play (aarch64) I installed the actual Entware distribution.
When compiling a source using 
`./configure --prefix=/opt`
I get an error
`checking for curl-config... no
configure: error: curl-config not found`
 
What have I to do to get curl-config installed ?
As I saw curl-config is part of libcurl since V 7.7.x. The acutal Entware distribution contains V 7.66.x, but curl-config isn't included in the /opt tree .

zyxmon, 2020-01-17:
We provide native compilation mostly for perl and python - users can build there own modules.
We do not provide support for native compilation.
You can build curl yoursel to get curl-config

closed 2020-01-17 by zyxmon

UserX404, 2021-01-10:
same problem here
Can anyone please give some information how to build curl with curl-config.
Thanks

zyxmon, 2021-01-10:
@UserX404 - make a patch to build without curl-config if possible or use crosscompiling.