Entware/Entware #394: Entware refuses to install

brendoncdrew, 2020-01-07:
Notice the line 43 and line 47 error's:
generic.sh: line 43: /opt/bin/opkg: not found
root@dd-wrt:/opt# ls -la /opt/bin
-rwxr-xr-x    0 root     root        166920 Jan  7 04:00 opkg

 however.... 

#!/bin/sh

TYPE='generic'
+ TYPE=generic
#TYPE='alternative'

unset LD_LIBRARY_PATH
+ unset LD_LIBRARY_PATH
unset LD_PRELOAD
+ unset LD_PRELOAD

echo 'Info: Checking for prerequisites and creating folders...'
+ echo 'Info: Checking for prerequisites and creating folders...'
if [ -d /opt ]; then
    echo 'Warning: Folder /opt exists!'
else
    mkdir /opt
fi
+ '[' -d /opt ]
+ echo 'Warning: Folder /opt exists!'
# no need to create many folders. entware-opt package creates most
for folder in bin etc lib/opkg tmp var/lock
do
  if [ -d "/opt/$folder" ]; then
    echo "Warning: Folder /opt/$folder exists!"
    echo 'Warning: If something goes wrong please clean /opt folder and try again.'
  else
    mkdir -p /opt/$folder
  fi
done
+ '[' -d /opt/bin ]
+ mkdir -p /opt/bin
+ '[' -d /opt/etc ]
+ mkdir -p /opt/etc
+ '[' -d /opt/lib/opkg ]
+ mkdir -p /opt/lib/opkg
+ '[' -d /opt/tmp ]
+ mkdir -p /opt/tmp
+ '[' -d /opt/var/lock ]
+ mkdir -p /opt/var/lock

echo 'Info: Opkg package manager deployment...'
+ echo 'Info: Opkg package manager deployment...'
DLOADER='ld.so.1'
+ DLOADER=ld.so.1
URL='http://bin.entware.net/mipselsf-k3.4/installer'
+ URL=http://bin.entware.net/mipselsf-k3.4/installer
wget $URL/opkg -O /opt/bin/opkg
+ wget http://bin.entware.net/mipselsf-k3.4/installer/opkg -O /opt/bin/opkg
Connecting to bin.entware.net (104.27.177.50:80)
saving to '/opt/bin/opkg'
'/opt/bin/opkg' saved
chmod 755 /opt/bin/opkg
+ chmod 755 /opt/bin/opkg
wget $URL/opkg.conf -O /opt/etc/opkg.conf
+ wget http://bin.entware.net/mipselsf-k3.4/installer/opkg.conf -O /opt/etc/opkg.conf
Connecting to bin.entware.net (104.27.176.50:80)
saving to '/opt/etc/opkg.conf'
'/opt/etc/opkg.conf' saved
wget $URL/ld-2.27.so -O /opt/lib/ld-2.27.so
+ wget http://bin.entware.net/mipselsf-k3.4/installer/ld-2.27.so -O /opt/lib/ld-2.27.so
Connecting to bin.entware.net (104.27.177.50:80)
saving to '/opt/lib/ld-2.27.so'
'/opt/lib/ld-2.27.so' saved
wget $URL/libc-2.27.so -O /opt/lib/libc-2.27.so
+ wget http://bin.entware.net/mipselsf-k3.4/installer/libc-2.27.so -O /opt/lib/libc-2.27.so
Connecting to bin.entware.net (104.27.176.50:80)
saving to '/opt/lib/libc-2.27.so'
'/opt/lib/libc-2.27.so' saved
wget $URL/libgcc_s.so.1 -O /opt/lib/libgcc_s.so.1
+ wget http://bin.entware.net/mipselsf-k3.4/installer/libgcc_s.so.1 -O /opt/lib/libgcc_s.so.1
Connecting to bin.entware.net (104.27.177.50:80)
saving to '/opt/lib/libgcc_s.so.1'
'/opt/lib/libgcc_s.so.1' saved
wget $URL/libpthread-2.27.so -O /opt/lib/libpthread-2.27.so
+ wget http://bin.entware.net/mipselsf-k3.4/installer/libpthread-2.27.so -O /opt/lib/libpthread-2.27.so
Connecting to bin.entware.net (104.27.176.50:80)
saving to '/opt/lib/libpthread-2.27.so'
'/opt/lib/libpthread-2.27.so' saved
cd /opt/lib
+ cd /opt/lib
chmod 755 ld-2.27.so
+ chmod 755 ld-2.27.so
ln -s ld-2.27.so $DLOADER
+ ln -s ld-2.27.so ld.so.1
ln -s libc-2.27.so libc.so.6
+ ln -s libc-2.27.so libc.so.6
ln -s libpthread-2.27.so libpthread.so.0
+ ln -s libpthread-2.27.so libpthread.so.0

echo 'Info: Basic packages installation...'
+ echo 'Info: Basic packages installation...'
/opt/bin/opkg update
+ /opt/bin/opkg update
generic.sh: line 43: /opt/bin/opkg: not found
if [ $TYPE = 'alternative' ]; then
  /opt/bin/opkg install busybox
fi
+ '[' generic '=' alternative ]
/opt/bin/opkg install entware-opt
+ /opt/bin/opkg install entware-opt
generic.sh: line 47: /opt/bin/opkg: not found

# Fix for multiuser environment
chmod 777 /opt/tmp
+ chmod 777 /opt/tmp

for file in passwd group shells shadow gshadow localtime; do
  if [ $TYPE = 'generic' ]; then
    if [ -f /etc/$file ]; then
      ln -sf /etc/$file /opt/etc/$file
    else
      [ -f /opt/etc/$file.1 ] && cp /opt/etc/$file.1 /opt/etc/$file
    fi
  else
    if [ -f /opt/etc/$file.1 ]; then
      cp /opt/etc/$file.1 /opt/etc/$file
    else
      [ -f /etc/$file ] && ln -sf /etc/$file /opt/etc/$file
    fi
  fi
done
+ '[' generic '=' generic ]
+ '[' -f /etc/passwd ]
+ ln -sf /etc/passwd /opt/etc/passwd
+ '[' generic '=' generic ]
+ '[' -f /etc/group ]
+ ln -sf /etc/group /opt/etc/group
+ '[' generic '=' generic ]
+ '[' -f /etc/shells ]
+ '[' -f /opt/etc/shells.1 ]
+ '[' generic '=' generic ]
+ '[' -f /etc/shadow ]
+ '[' -f /opt/etc/shadow.1 ]
+ '[' generic '=' generic ]
+ '[' -f /etc/gshadow ]
+ '[' -f /opt/etc/gshadow.1 ]
+ '[' generic '=' generic ]
+ '[' -f /etc/localtime ]
+ '[' -f /opt/etc/localtime.1 ]

echo 'Info: Congratulations!'
+ echo 'Info: Congratulations!'
echo 'Info: If there are no errors above then Entware was successfully initialized.'
+ echo 'Info: If there are no errors above then Entware was successfully initialized.'
echo 'Info: Add /opt/bin & /opt/sbin to $PATH variable'
+ echo 'Info: Add /opt/bin & /opt/sbin to $PATH variable'
echo 'Info: Add "/opt/etc/init.d/rc.unslung start" to startup script for Entware services to start'
+ echo 'Info: Add "/opt/etc/init.d/rc.unslung start" to startup script for Entware services to start'
if [ $TYPE = 'alternative' ]; then
  echo 'Info: Use ssh server from Entware for better compatibility.'
fi
+ '[' generic '=' alternative ]
echo 'Info: Found a Bug? Please report at https://github.com/Entware/Entware/issues'
+ echo 'Info: Found a Bug? Please report at https://github.com/Entware/Entware/issues'

so here's some info about my system:

Iam running dd-wrt latest firmware 1/05/2020:
Router Model Linksys E1200 v2
Firmware Version DD-WRT v3.0-r41892 mega (01/05/20)
Kernel Version Linux 3.10.108-d8 #1348 Sun Jan 5 13:46:22 +04 2020 mips

also:
root@dd-wrt:/opt# export
export HOME='/tmp/root'
export LD_LIBRARY_PATH='/lib:/usr/lib:/jffs/lib:/jffs/usr/lib:/jffs/usr/local/lib:/mmc/lib:/mmc/usr/lib:/opt/lib:/opt/usr/lib'
export LOGNAME='root'
export OLDPWD='/tmp'
export PATH='/bin:/usr/bin:/sbin:/usr/sbin:/jffs/sbin:/jffs/bin:/jffs/usr/sbin:/jffs/usr/bin:/mmc/sbin:/mmc/bin:/mmc/usr/sbin:/mmc/usr/bin:/opt/sbin:/opt/bin:/opt/usr/sbin:/opt/usr/bin'

All the files seem to have installed correctly and exist where they should be at howeve when i try to run opkg.... 

root@dd-wrt:/opt# /opt/bin/opkg
-sh: /opt/bin/opkg: not found

Wtf???

Any help would be appreciated...

zyxmon, 2020-01-07:
Probably wrong architecture was selected.
We do not provide this sort of support. Ask for support at DD-WRT forum.

closed 2020-01-07 by zyxmon

brendoncdrew, 2020-01-07:
Well since I didnt even have tge option to select anything like that i
doubt it... Your install scripy did all that on its own so maybe you should
fix that