Entware/Entware #397: Update mpd to 0.21.18 version

AndriiZ, 2020-01-14:
Package request

mpd in repo has a version 0.20.23-3
---
mpd in 0.21 branch added few usefull features, for example forcing charset for icecast streams with cp1251 header. Existing version can not properly decode header in cp1251, while 0.21 can

- [+ ] No similar package already added and/or this one got some new killer feature(s),
- [+] Package is in active development (at least, not abandoned for 5 years or more),
- [+] I'm ready to test added package on my {platform} and get a feedback about it.

---

AndriiZ, 2020-01-14:
https://github.com/MusicPlayerDaemon/MPD/issues/616

ryzhovau, 2020-02-04:
It's up to upstream. We synced recently, mpd updated to 0.21.16.

closed 2020-02-04 by ryzhovau