Entware/Entware #400: Postfix on QNAP

LincDavis, 2020-01-20:
There are some problems with the Postfix package that keep it from working as is under QTS.

1. In the file /opt/etc/postfix/postfix-files, all references to user 'root' should be changed to user 'admin'.

2. In the file /opt/libexec/postfix/postfix-script, references to user 'root' should be changed to user 0 or 'admin'.

3. Missing symbolic links should be created:

```
ln -s sendmail.postfix /opt/sbin/sendmail
ln -fs ../sbin/sendmail.postfix /opt/bin/newaliases.postfix
```

4. The package has a hidden dependency on GNU find, which must be installed and its location added to the PATH in the scripts.

5. User 'postfix' and group 'postdrop' should be created:

```
adduser -H -h /opt/var/spool/postfix -D -s /bin/false postfix
addgroup postdrop
```

6. Secure relaying doesn't work because the binary doesn't look for the Cyrus SASL libraries where the package installs them. I work around this by linking back to them:

`ln -s /opt/lib/sasl2 /usr/lib/`

This has to be redone by a cron job at boot time because the link is deleted.

7. In master.cf, the line

`submission inet n       -       n       -       -       smtpd`

should be changed to

`587         inet n       -       n       -       -       smtpd`

This is necessary because of the simplified /etc/services file in QTS.

zyxmon, 2020-01-20:
One can use `alternative` qnap package - in this case Entware superuser will be root (qnap firware will use /etc/passwd and Entware will use /opt/etc/passwd). find (findutils package) is a part of initial optware installation. As for user 'postfix' and group 'postdrop' - we do not add users in postinst scripts because it is firmware dependent. Entware is used on many differen devices (routers, NASes, android boxes, etc). We will consider other problems you've found.

zyxmon, 2020-02-01:
Most of devices, that use Entware have readonly file system. Symlinks cannot be created.

closed 2020-02-01 by zyxmon

LincDavis, 2020-02-02:
There would be no need for a symlink outside of `/opt` if the library search path was correct.

The Postfix `post-install` script doesn't work because the PATH is wrong. It should be something like this so that GNU find is called instead of BusyBox find:

`PATH=/opt/bin:/bin:/usr/bin:/usr/sbin:/usr/etc:/sbin:/etc`

The script `postfix-script` also needs a PATH assignment, and on lines 297, 314, and 361, the find predicate `-user root` should be `-user 0` or `-user admin`.

zyxmon, 2020-02-03:
PATH variable is set in /opt/etc/profile and starts with /opt.... for interactive session and is set in /opt/etc/init.d/rc.unslung that starts services.
There is no post-install script for postfix package.
Please make a PR at GitHub to fix `postfix-script`.