Entware/Entware #401: Package update request: php7-mod-tokenizer

indigob0y, 2020-01-20:
Please update tokenizer module for php. Current version doesn't work.

`uname -a`
Linux HQ 4.4.187 #650 SMP PREEMPT Tue Aug 6 11:38:46 +04 2019 armv7l DD-WRT

Steps to reproduce:

`opkg install php7-mod-tokenizer`
`php`
`<?php dl('tokenizer') ?>`

output:
Warning: dl(): tokenizer: Unable to initialize module
Module compiled with module API=20170718
PHP    compiled with module API=20180731
These options need to match
 in Standard input code on line 1

ryzhovau, 2020-02-04:
Can't reproduce it on fresh install.
We synced php7 with upstream recently. If `opkg upgrade` wont help, please, install Entware from scratch.

closed 2020-02-04 by ryzhovau

indigob0y, 2020-02-06:
Thank you for that comment, it helped me solve the problem. I didn't notice an older version of php was installed in `/usr/bin`. I configured `lighttpd` to use the one in `/opt/bin` and now it is running correctly.