Entware/Entware #403: Package request: pinentry

vimanderer934, 2020-01-20:
Package request: https://www.gnupg.org/software/pinentry/index.html
---
pinentry is a small collection of dialog programs that allow GnuPG to read passphrases and PIN numbers in a secure manner. 

This is necessary for GPG integration with Vim and other interactive programs.

- [x ] No similar package already added and/or this one got some new killer feature(s),
- [x ] Package is in active development (at least, not abandoned for 5 years or more),
- [ x] I'm ready to test added package on my {platform} and get a feedback about it.

vimanderer934, 2020-04-05:
I created a Makefile that successfully compiles this package.

The pull request for rtndev is here if this is appropriate:
https://github.com/Entware/rtndev/pull/13

ryzhovau, 2020-08-04:
Thanks for PR again.

closed 2020-08-04 by ryzhovau