Entware/Entware #407: "make menuconfig" fails because hostapd package caanot find wpad package

egorenar, 2020-02-01:
Tested with armv7-3.2 config.

WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-mesh-openssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-mesh-wolfssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-basic', which does `

ryzhovau, 2020-02-01:
You may ignore this messages. As you can see from `make menuconfig`, not all upstream packages are used.

closed 2020-02-01 by ryzhovau

zyxmon, 2020-02-01:
Please note - we cannot help with packages that are not part of Entware. hostapd is not in Entware.

egorenar, 2020-02-01:
hmm, i would like to ignore it but i get a syntax error in tmp/.config-packages.in.

zyxmon, 2020-02-01:
The instructions in wiki are valid for debian 9 and 10 x64. What is your distro?
Try to clean tmp subfolder and remove hostapd folder.

egorenar, 2020-02-01:
My distro is void linux and it worked fine in december 2019.
I cleaned everything with git clean, removing packages/network/services/hostapd helps but i consider it a hack.

egorenar, 2020-02-01:
It works on Debian, sorry for spamming :)