Entware/Entware #410: onmp_bug

jess2020github, 2020-02-05:
mipssf-k3.4 中 php7.4 
但是 ownclound (nextclound)需要 <php7.2
why?

ryzhovau, 2020-02-06:
In English, please.

zyxmon, 2020-02-06:
Probably this will help - https://forum.keenetic.net/topic/3942-owncloud-%E2%80%94-the-private-cloud

ryzhovau, 2020-03-04:
No response.

closed 2020-03-04 by ryzhovau

jess2020github, 2020-03-05:
now  in onmp    nextcloud  not run <PHP7.3,but  entware mipself 3.4  with PHP7.4
but i'm backup entware  mipself 3.2 ,is runnig onmp no problem, now   http://bin.entware.net   not included mipself 3.2

| |
汪军
|
|
tjmy_wj@126.com
|
签名由网易邮箱大师定制

xiangfeigao, 2020-04-25:
能否把备份的发给我一份

jess2020github, 2020-04-25:
文件比较大，最近我在安装黑群晖，备份文件暂时不好找，等过几天我找找看，压缩后比较小，可以发邮件，如果比较大，我就给你百度网盘共享一下

| |
汪军
|
|
tjmy_wj@126.com
|
签名由网易邮箱大师定制

能否把备份的发给我一份

jess2020github, 2020-04-26:
我只有         mipselsf-k3.2和mipselsf-k3.4，其中mipselsf-k3.2 （大小750M），自带的PHP<7.3, mipselsf-k3.4(大小2.2G），我现在出差，明天到家后上传到百度网盘，共享给你。
| |
汪军
|
|
tjmy_wj@126.com
|
签名由网易邮箱大师定制
On 4/26/2020 05:50，汪军<tjmy_wj@126.com> wrote：
文件比较大，最近我在安装黑群晖，备份文件暂时不好找，等过几天我找找看，压缩后比较小，可以发邮件，如果比较大，我就给你百度网盘共享一下

| |
汪军
|
|
tjmy_wj@126.com
|
签名由网易邮箱大师定制

xiangfeigao, 2020-04-26:
谢谢，我只想用padavan搭建nextcloud，版本不限

ryzhovau, 2020-04-26:
In English, please. 
Guys, keep in mind Google translate or similar tools can't get appropriate translation of your discussion.

jess2020github, 2020-04-27:
https://pan.baidu.com/s/1j7hNjX4EH2Jqxaknf_iWSg
password:o9bw

include: onmp LIB  (mipselsf-k3.2.rar   <php7.3    test OK)
                mipselsf-k3.4rar    php7.4 ,don't run ownclude
            

| |
汪军
|
|
tjmy_wj@126.com
|
签名由网易邮箱大师定制