Entware/Entware #412: cryfs dies with `Illegal instruction`

greyltc, 2020-02-09:
I have an old QNAP TS-419U+ with
```
# cat /proc/cpuinfo
Processor name	: Feroceon 88F6282 rev 1 (v5l) @ 1.6 GHz 
BogoMIPS	: 1587.60
Features	: swp half thumb fastmult edsp 
CPU implementer	: 0x56
CPU architecture: 5TE
CPU variant	: 0x2
CPU part	: 0x131
CPU revision	: 1

Hardware	: Feroceon-KW ARM
Revision	: 0000
Serial		: 0000000000000000
```
and
```
# uname -a                                                                                                       
Linux redacted 3.4.6 #1 Thu Jan 9 09:44:32 CST 2020 armv5tel unknown
```
I've installed the cryfs from your repo
```
# opkg list-installed | grep cryfs 
cryfs - 0.10.2-1
```
but when I try to run it, it dies after some moments with an "Illegal instruction" message
```
# cryfs
Illegal instruction
```
Running it with gdb gives a hint about why it dies
```
# gdb cryfs
GNU gdb (GDB) 8.3.1
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-openwrt-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from cryfs...
(No debugging symbols found in cryfs)
(gdb) run
Starting program: /share/MD0_DATA/.qpkg/Entware/bin/cryfs 
warning: File "/share/MD0_DATA/.qpkg/Entware/lib/libthread_db-1.0.so" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
To enable execution of this file add
	add-auto-load-safe-path /share/MD0_DATA/.qpkg/Entware/lib/libthread_db-1.0.so
line to your configuration file "/root/.gdbinit".
To completely disable this security protection add
	set auto-load safe-path /
line to your configuration file "/root/.gdbinit".
For more information about this security protection see the
"Auto-loading safe path" section in the GDB manual.  E.g., run from the shell:
	info "(gdb)Auto-loading safe path"
warning: Unable to find libthread_db matching inferior's thread library, thread debugging will not be available.

Program received signal SIGILL, Illegal instruction.
0x0032535c in CryptoPP::CPU_ProbeARMv7() ()
```
So at this point, I'm not sure if this is an upstream bug in https://github.com/weidai11/cryptopp or if this is a build configuration issue.

Any help here is welcome!

greyltc, 2020-02-09:
Apparently this can be fixed by setting `#define CRYPTOPP_DISABLE_ASM 1` in config_asm.h before compiling cryptopp.

noloader, 2020-02-09:
Yes, `#define CRYPTOPP_DISABLE_ASM 1` in `config_asm.h` is the safest option. It ensures the library and programs linking to the library get the same option.

You can also add `-DCRYPTOPP_DISABLE_ASM=1` to `CXXFLAGS`. But that is trickier because it requires all programs linking to Crypto++ to use the same `CXXFLAGS`. That's a problem equivalent to RTFM, and it will never work. (Or it has not worked in the last 50 years. I doubt it will start working now).

greyltc, 2020-02-09:
I'm struggling to figure out exactly where cryptopp is even being built here so that I can figure out when and where to patch its config_asm.h ...

zyxmon, 2020-02-09:
cryptopp is not used in Entware.
bug in cryfs confirmed!

greyltc, 2020-02-09:
gdb shows the Illegal instruction is in `CryptoPP::CPU_ProbeARMv7()`. How can Entware's cryfs not be using it?

zyxmon, 2020-02-09:
cryptopp is in vendor subdir of cryfs sources.
Here is a fixed binary - http://test.entware.net/armv5sf-k3.2/cryfs
Please check if it works and report.

greyltc, 2020-02-09:
```
# ./cryfs --version
CryFS Version 0.10.2
```
Much better! Thank you. It at least runs.
It'll take me a bit to verify more of its functionality, but this is a great start.

ryzhovau, 2020-02-27:
Done. https://github.com/Entware/rtndev/commit/cfff77aefdc07508fab459d0c4d1330a08d361b2

closed 2020-02-27 by ryzhovau