Entware/Entware #415: Package Request : Nebula

richardjchadwick, 2020-02-12:
Package request for nebula https://github.com/slackhq/nebula
---
A scalable overlay networking tool with a focus on performance, simplicity and security

- [x] No similar package already added and this one got some new killer feature(s),
- [ x] Package is in active development
- [x] I'm ready to test added package on my {armv7 -nohp} and give a feedback about it.
 My arm7 -nohp platform is actually an ASUS RT-AC66U running linux 4.4 

Tried to build myself but was unsuccessful but I have no experience cross-compiling, or embedded development or with Openwrt or entware development.

pasha-zzz, 2020-02-13:
Its written in Go...
https://github.com/Entware/Entware/issues/329#issuecomment-530288541

zyxmon, 2020-02-15:
@richardjchadwick - I can give you a package to test. What package architecture do you need?

zyxmon, 2020-02-15:
http://bin.entware.net/armv7sf-k3.2/test/nebula_nohf_1.1.0-1_armv7-3.2.ipk
This may be your arch

richardjchadwick, 2020-02-15:
That's correct.  Won't be able to test until Tuesday. Will let you know and
THANK YOU.

zyxmon, 2020-02-23:
closed via https://github.com/Entware/entware-go/commit/5c7600e20cf0f45f49bb2959056d40cdcd6f3de9

If anybody wants to test nebula plz write what architecture you use.

closed 2020-02-23 by zyxmon