Entware/Entware #416: Synology error armv5

anasaln, 2020-02-12:
When installing Entware on a synology ds211+ the result is : "kernel toot old".

# uname -m
armv5tel

wget -O - http://bin.entware.net/armv5sf-k3.2/installer/generic.sh | /bin/sh

......
Info: Basic packages installation...
FATAL: kernel too old
FATAL: kernel too old
Info: Congratulations!
Info: If there are no errors above then Entware was successfully initialized.
Info: Add /opt/bin & /opt/sbin to your PATH variable
Info: Add '/opt/etc/init.d/rc.unslung start' to startup script for Entware services to start
Info: Found a Bug? Please report at https://github.com/Entware/Entware/issues

anasaln, 2020-02-12:
Sorry
I have only read material ref !
My dsm is too old
# uname -r
2.6.32.12

But no updates for 211+ now  so no Entware

closed 2020-02-12 by anasaln

ryzhovau, 2020-02-13:
> But no updates for 211+ now so no Entware

Take a look at [Entware-ng](http://pkg.entware.net/binaries/armv5/). It's obsolete, but still may fit your needs.