Entware/Entware #418: libjson-c

jjsarton, 2020-02-15:
I have wrote a little application which shall use /opt/lib/libjson-c.so.4

If I compile the application the library which is searches is libjson-c.so.2 so I get an error on QNAP
If I add the link /opt/lib/libjson-c.so -> libjson-c.so.4 the binary bind the correct library.
Please provides  the links lib*.so are provided.

ryzhovau, 2020-02-15:
Won't fix that until it used by other packages from repo, sorry.

closed 2020-02-15 by ryzhovau

jjsarton, 2020-02-15:
Hello, first I respect your decision.
I have wrote a little application in order to provide DNS resolution. At this time this is running on an RaspberryPI and since I have two NAS (Synology and QNAP) I have compiled the application on both systems. I use the curl and json-c libraries which are also provided by entware. I have compiled the binary on the NAS systems. The first problem what that I don't had the corresponding include files on the NAS systems and the second the problem with the wrong library.
Before I put the modified sources on Github I want to know if there is a possibility to get the corresponding include files from  entware and if it is possible to provide the sources without the need to use workarounds.