Entware/Entware #424: [Warning] InnoDB: Failed to set memory to MADV_DONTDUMP: Function not implemented

ATAGAU, 2020-02-23:
A month ago the Mariadb server and client worked perfectly with phpMyAdmin, but since I reinstalled the entire system I have appeared in the SSH console:

/opt/bin/mysql_install_db --user=admin --basedir=/opt --datadir=/opt/var/mysql/ --lc-messages-dir=/opt/share/mariadb

Installing MariaDB/MySQL system tables in '/opt/var/mysql/' ...
2020-02-15 18:55:02 0 [Warning] InnoDB: Failed to set memory to MADV_DONTDUMP: F unction not implemented ptr 0x7f73bff000 size 134217728
2020-02-15 18:55:02 0 [Warning] InnoDB: Failed to set memory to MADV_DONTDUMP: F unction not implemented ptr 0x7f71bff000 size 33554432
2020-02-15 18:55:02 0 [Warning] InnoDB: Failed to set memory to MADV_DONTDUMP: Function not implemented ptr 0x7f8014a000 size 2097152
2020-02-15 18:55:02 0 [Warning] InnoDB: Failed to set memory to MADV_DODUMP: Function not implemented ptr 0x7f8014a000 size 2097152
2020-02-15 18:55:03 0 [Warning] InnoDB: Failed to set memory to MADV_DODUMP: Function not implemented ptr 0x7f71bff000 size 33554432
2020-02-15 18:55:03 0 [Warning] InnoDB: Failed to set memory to MADV_DODUMP: Function not implemented ptr 0x7f73bff000 size 134217728
OK

Impossible to start the server with the command:

/opt/etc/init.d/S70mysqld start

Is it possible to deactivate the MADV_DODUMP function ?. The "-core-file" and "-innodb-buffer-pool-in-core-file" commands do not exist in this system Linux.

I have a previous backup and this does not happen. This error occurs when you upgrade from 10.2.26 to 10.4.10 or directly installing version 10.4.10

zyxmon, 2020-02-23:
This is just a warning

ATAGAU, 2020-02-23:
But I don't know how to avoid it, it doesn't start, it doesn't let you do anything

zyxmon, 2020-02-23:
mariadb was updated (shebang fixes). It may help or may be no,

ATAGAU, 2020-02-24:
I have updated the firmware to 384.15, I have installed entware from amtm and now if it starts mysqld, but I get the following errors:

/opt/bin/mysqladmin -u root password 'newpass'
/opt/bin/mysqladmin: connect to server at 'localhost' failed
error: 'Access denied for user 'root'@'localhost''

and

mysql -u root -p
Enter password:
ERROR 1698 (28000): Access denied for user 'root'@'localhost'

I can't use the "sudo" command, this is a router ....

zyxmon, 2020-02-24:
We have reports that mariadb works on routers with root (not admin) as superuser.
You are to find the solution yourself.

closed 2020-02-24 by zyxmon