Entware/Entware #426: Package request: XFS kernel module

Anuskuss, 2020-03-01:
Enable the use of storage devices with the XFS file system.

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my {platform} and get a feedback about it.

I did some research and I don't think the repo has this yet.

zyxmon, 2020-03-01:
There are no kernel modules in Entware. It is impossible.

closed 2020-03-01 by zyxmon

Anuskuss, 2020-03-01:
Yeah I get that it's unfeasible but certainly not impossible, right? IIRC optware had `kmod-xfs`/`kmod-fs-xfs`.

zyxmon, 2020-03-02:
If you have kernel modules from any source - you can use them with Entware binaries.
To build kernel modules one needs kernel sources with patches for the device you are using and kernel configuration of the device firmware.
Entware uses vanilla kernel sources with default kernel configuration for a wide range of devices. Kernel modules compiled in Entware will not load in 99.99999% cases.

Anuskuss, 2020-03-02:
Well I gave it a shot and tried to compile it myself but I'm stuck. I'm using OpenWRT's [kmod-fs-xfs](https://archive.openwrt.org/releases/17.01.6/targets/bcm53xx/generic/packages/kmod-fs-xfs_4.4.153-1_arm_cortex-a9.ipk) as a base but I can't load the module. `dmesg` shows
```
xfs: Unknown symbol warn_slowpath_null (err 0)
```
Do you have any idea how I can solve that? I know this is not strictly related to Entware but I feel like you may know what's wrong.

zyxmon, 2020-03-02:
You need kernel sources, kernel config and preferable toolchain from firmware developer to compile module.

Anuskuss, 2020-03-02:
I have the OpenWrt build system all set-up. My router (Netgear R6400v1) is pretty much identical to Netgear R6300v2 so I'm assuming since they can create packages for that router, I should be able to create them for mine as well, shouldn't I? What am I missing?

zyxmon, 2020-03-02:
> I should be able to create them for mine as well, shouldn't I? 

Only in case the kernel sources and config are for the exact firmware version of you device (not a similar one)!