Entware/Entware #429: telnet client

bulbalion, 2020-03-08:
Package request
---
I would be very glad if the old telnet would be supported by entware. I'm currently using my NAS box as a BBS with entware and telnet would be exactly what I would need very much.

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my {platform} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

ryzhovau, 2020-03-08:
You may use `telnet` applet from busybox.

bulbalion, 2020-03-08:
Very well... thank You for Your fast and nice reply.

closed 2020-03-08 by bulbalion