Entware/Entware #430: Package request: buildtorrent

tomyang001, 2020-03-08:
Package request
---
buildtorrent is a torrent file creation program. Given an announce url and an input file or directory, buildtorrent generates an output .torrent file that can be used by torrent clients. It would be much easier using this to create torrent files remotely through ssh than create them locally after transferring the input files.
**[https://mathr.co.uk/blog/2010-01-31_buildtorrent-0.8_released.html](https://mathr.co.uk/blog/2010-01-31_buildtorrent-0.8_released.html)**

- [x] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my {armv7-3.2} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

ryzhovau, 2020-03-09:
What's the killer feature of `buildtorrent` comparing to `ctorrent` or `transmission-create`?

tomyang001, 2020-03-09:
The name "buildtorrent" obviously suggests it can create torrents files, the function is quite simple and the size of its binary should only be around 20 KB. I didn't know `ctorrent` could create torrent files, it seems a program mainly used to download torrents. If I need to get `transmission-create` I need to install `transmission-cli-openssl`, there isn't a package named "transmission-create".

For the dependencies, `buildtorrent` should only depend on libc, much simpler than the other two.

ryzhovau, 2020-03-09:
I see no sense to add several tools with the same functions, sorry.

closed 2020-03-09 by ryzhovau