Entware/Entware #431: Update qBittorrent

magpte, 2020-03-13:
Update: https://github.com/qbittorrent/qBittorrent

qBittorrent is a bittorrent client programmed in C++ / Qt that uses libtorrent (sometimes called libtorrent-rasterbar) by Arvid Norberg.

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my {MIPS/Pavdavan} and get a feedback about it

giovEra, 2020-07-18:
**+1** - the latest version of qBittorrent has a much improved Web Interface (it is amazing).

Alexey71, 2020-09-03:
Please update

NickSniper, 2020-09-10:
Please, please update to the latest version. It is a lot of "sweet" changes.

ryzhovau, 2020-09-11:
Done. Binaries will arrive in next week.

closed 2020-09-11 by ryzhovau

DanielSmedegaardBuus, 2021-03-05:
Is this in some special repo? Or... I'm totally new to entware and packages on my Asus router :D Used to just "normal" computers and Debian, Ubuntu, and so on... Can't find qbittorrent in my `opkg list` list... I see it here `http://bin.entware.net/armv7sf-k3.2/Packages.html` — but don't know ...

zyxmon, 2021-03-05:
No chance to compile Qt for old kernels!

Alexey71, 2021-03-14:
Hello. You can update to the latest version?, please

Alexey71, 2021-04-18:
@ryzhovau Александр здравствуйте. Можете пожалуйста обновить до последней версии, не хватает фишки "Размещение содержимого торрента", она есть в новых версиях только...

ryzhovau, 2021-04-18:
Пока не могу из-за версии rasterbar-libtorrent, [проверял](147267665) пару дней назад.