Entware/Entware #437: Gerbera compile options

pasha-zzz, 2020-03-18:
Especially  needed PROTOCOL_EXTENSIONS. W/o it subj dont add custom headers and (f.e.) Samsung TVs say "Unsupported file format".

Also why it compiled with debug?

Compile info
-------------
WITH_MAGIC=1
WITH_MYSQL=0
WITH_CURL=1
WITH_INOTIFY=1
WITH_JS=1
WITH_TAGLIB=0
WITH_AVCODEC=1
**WITH_FFMPEGTHUMBNAILER=0**
WITH_EXIF=1
**WITH_EXIV2=0**
**WITH_PROTOCOL_EXTENSIONS=**
WITH_SYSTEMD=0
WITH_LASTFM=0
**WITH_DEBUG=1**
WITH_TESTS=0

ryzhovau, 2020-03-18:
The `mediatomb`, a ancestor of `gerbera` was a quite fragile thing and database corruption was appeared too often. Debug info still present in binaries since that times.

I'll add other valuable options if possible.

ryzhovau, 2020-03-22:
> Especially needed PROTOCOL_EXTENSIONS. W/o it subj dont add custom headers and (f.e.) Samsung TVs say "Unsupported file format".

Please, tell me where did you get this info? ` PROTOCOL_EXTENSIONS` definition [is not used](https://github.com/gerbera/gerbera/search?q=WITH_PROTOCOL_EXTENSIONS) in gerbera sources.

pasha-zzz, 2020-03-24:
https://github.com/gerbera/gerbera/issues/352

closed 2020-08-06 by The-BB