Entware/Entware #439: Package request: fail2ban

bugsyb, 2020-03-18:
Package request
---
Fail2Ban: ban hosts that cause multiple authentication errors
https://github.com/fail2ban/fail2ban

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my {platform} and get a feedback about it.

---
Getting it ported to Entware shouldn't be overly complicated.

### taken from github
Required:
    Python2 >= 2.6 or Python >= 3.2 or PyPy
Optional:
    pyinotify >= 0.8.3, may require:
        Linux >= 2.6.13
    gamin >= 0.0.21
    systemd >= 204 and python bindings:
        python-systemd package
    dnspython

To install:
tar xvfj fail2ban-1.0.1.tar.bz2
cd fail2ban-1.0.1
sudo python setup.py install

Alternatively, you can clone the source from GitHub to a directory of Your choice, and do the install from there. Pick the correct branch, for example, master or 0.11

git clone https://github.com/fail2ban/fail2ban.git
cd fail2ban
sudo python setup.py install 

This will install Fail2Ban into the python library directory. The executable scripts are placed into /usr/bin, and configuration in /etc/fail2ban.

ryzhovau, 2020-04-19:
We don't pack `fail2pack` because it's too heavy for most of embedded devices.
You can always install it from sources.

bugsyb, 2020-04-19:
Embedded devices are quite powerful these days, i.e. Entware used on Odroid N2, just to name one.
Installing from sources takes the complexity and number of packages to be installed to the next level.
It would be kind if it could be packaged anyway, please.
I'd be keen to contribute too, if only could get some pointers re howto/rtfm.

Thanks.

ryzhovau, 2020-04-19:
I'm agree Odroid N2 is powerful, still it's an embedded device as well. I'm afraid there's no clear decision which device can be named as powerful and which is not. Entware primary purpose was to enhance device limited firmware functions. In the other hand, Odroid/Rpi/etc boards powerful enough to run full blown Linux disto(s) and doesn't need Entware in addition.

Please, don't presume my previous answer as "no, because I said no":) In common, `fail2ban` is package for some router and most of routers are quite weak MIPS/ARM devices. I've tried it several years ago on 128MB RAM MIPS router and decided to use more light-weigh solutions then `fail2ban`.

If you are using full blown Linux on Odroid N2 and if it's a router, may be it's better to install `fail2ban` there.

Please, feel free to describe you `fail2ban` use case, may be I just lagged behind modern life.

closed 2020-05-10 by no-response[bot]