Entware/Entware #440: GCC not work

rampageX, 2020-03-26:
Entware running on Netgear R7000 with Tomato firmware:

`Linux netgear 2.6.36.4brcmarm #7 SMP PREEMPT Thu Mar 19 17:12:50 CET 2020 armv7l GNU/Linux`

I followed `https://github.com/Entware/Entware/wiki/Using-GCC-for-native-compilation` to build some app, but not work even `gcc -v`:

```
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/lto-wrapper
Target: arm-openwrt-linux-gnueabi
Configured with: /media/ware3/Entware.2020.01/build_dir/target-arm_cortex-a9_glibc-2.23_eabi/gcc-7.4.0/configure --target=arm-openwrt-linux --host=arm-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix= --program-suffix= --prefix=/opt --exec-prefix=/opt --bindir=/opt/bin --sbindir=/opt/sbin --libexecdir=/opt/lib --sysconfdir=/opt/etc --datadir=/opt/share --localstatedir=/opt/var --mandir=/opt/man --infodir=/opt/info --disable-nls 'CXXFLAGS_FOR_TARGET=-g -O2 -D_GLIBCXX_INCLUDE_NEXT_C_HEADERS' --build=x86_64-pc-linux-gnu --host=arm-openwrt-linux-gnueabi --target=arm-openwrt-linux-gnueabi --enable-languages=c,c++ --with-bugurl=https://dev.openwrt.org/ --with-pkgversion='OpenWrt GCC 7.4.0' --enable-shared --enable-__cxa_atexit --with-default-libstdcxx-abi=gcc4-compatible --enable-target-optspace --with-gnu-ld --with-specs='-Wl,-rpath=/opt/lib -Wl,--dynamic-linker=/opt/lib/ld-linux.so.3' --with-boot-ldflags='static-libstdc++ -static-libgcc -Wl,-rpath=/opt/lib -Wl,--dynamic-linker=/opt/lib/ld-linux.so.3' --disable-nls --disable-libsanitizer --disable-libvtv --disable-libcilkrts --disable-libmudflap --disable-libmpx --disable-multilib --disable-libgomp --disable-libquadmath --disable-libssp --disable-decimal-float --disable-libstdcxx-pch --with-host-libstdcxx=-lstdc++ --prefix=/opt --libexecdir=/opt/lib --with-local-prefix=/opt --with-stage1-ldflags='-lstdc++ -Wl,-rpath=/opt/lib -Wl,--dynamic-linker=/opt/lib/ld-linux.so.3' --with-float=soft --with-default-libstdcxx-abi=gcc4-compatible
Thread model: posix
gcc version 7.4.0 (OpenWrt GCC 7.4.0)
COMPILER_PATH=/opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/:/opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/:/opt/lib/gcc/arm-openwrt-linux-gnueabi/:/opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/:/opt/lib/gcc/arm-openwrt-linux-gnueabi/
LIBRARY_PATH=/opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/:/opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-mfloat-abi=soft' '-mtls-dialect=gnu'
 /opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/collect2 -plugin /opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/liblto_plugin.so -plugin-opt=/opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/lto-wrapper -plugin-opt=-fresolution=/mmc/tmp/cc0iY3UN.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s --eh-frame-hdr -dynamic-linker /lib/ld-linux.so.3 -X -m armelf_linux_eabi /opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/crt1.o /opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/crti.o /opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/crtbegin.o -L/opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0 -L/opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/../../.. -rpath=/opt/lib --dynamic-linker=/opt/lib/ld-linux.so.3 -lgcc_s -lc -lgcc_s /opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/crtend.o /opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/crtn.o
/opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/crt1.o: In function `_start':
(.text+0x34): undefined reference to `main'
collect2: error: ld returned 1 exit status
```

Simple app `microsocks` build:

```
gcc -Wl,-rpath=/opt/lib -Wl,--dynamic-linker=/opt/lib/ld-linux.so.3 -L/opt/lib sockssrv.o server.o sblist.o sblist_delete.o -lpthread -o microsocks
/opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/crt1.o: In function `_start':
(.text+0x30): undefined reference to `__libc_csu_fini'
/opt/lib/gcc/arm-openwrt-linux-gnueabi/7.4.0/crt1.o: In function `_start':
(.text+0x38): undefined reference to `__libc_csu_init'
collect2: error: ld returned 1 exit status
make: *** [Makefile:31: microsocks] Error 1
```
My `env`:

```
SSH_CLIENT=192.168.2.8 52533 22
USER=root
CGO_ENABLED=0
SHLVL=1
OLDPWD=/opt/etc
TERMINFO=/opt/share/terminfo
HOME=/opt/etc/home
SSH_TTY=/dev/pts/0
LDFLAGS=-Wl,-rpath=/opt/lib -Wl,--dynamic-linker=/opt/lib/ld-linux.so.3 -L/opt/lib
PS1=\[\033[0;31m\][entware]\[\033[0;37m\][\[\033[0;32m\]\h\[\033[0;37m\]:\[\033[0;36m\]\w\[\033[0;37m\]]$ \[\033[0m\]
TMPDIR=/mmc/tmp
LOGNAME=root
TEMP=/opt/tmp
TERM=xterm-color
GIT_SSL_NO_VERIFY=true
PATH=/opt/bin:/opt/sbin:/opt/usr/bin:/opt/usr/sbin:/bin:/sbin:/usr/bin:/usr/sbin:/usr/local/sbin:/opt/etc/init.d
CFLAGS=-O2 -pipe -march=armv7-a -mtune=cortex-a9 -fno-caller-saves -mfloat-abi=soft
TMP=/opt/tmp
GOPATH=/mnt/data/compile/go
SHELL=/bin/sh
GCC_COLORS=error=01;31:warning=01;35:note=01;36:caret=01;32:locus=01:quote=01
PWD=/mnt/data/compile/microsocks
SSH_CONNECTION=192.168.2.8 52533 192.168.2.1 22
M4=/opt/bin/m4
CC=gcc
TZ=CST-8
PKG_CONFIG_LIBDIR=/opt/lib/pkgconfig
```

zyxmon, 2020-04-26:
gcc is experimental in Entware. Not all projects can be compiled with it. Try to google for the solution. May be changing ld to gcc to link binary will help.

closed 2020-04-26 by zyxmon

ellisonpatterson, 2021-02-23:
Same issue, did changing ld solve this?

zyxmon, 2021-02-24:
@ellisonpatterson - try to add `libc_nonshared.a` when linkung.