Entware/Entware #442: Package Request : mtd-utils, mkfs.jffs2

M95D, 2020-03-30:
Package request
---
mtd-utils, especially mkfs.jffs2 is indispensable tool for routers and other embedded devices.

- [x] No similar package already added,
- [x] Package is in active development,
- [x] I'm ready to test added package on my router (arm-brcm) and get a feedback about it.

Thank you.

KyleSanderson, 2020-05-03:
@ryzhovau any update on this one? I'm in quite a jam and am quite surprised https://github.com/Entware/Entware/tree/master/package/utils/mtd-utils is in the tree but not built...

M95D, 2020-05-05:
I had to compile it myself from the [official sources](http://linux-mtd.infradead.org/), not entware.

KyleSanderson, 2020-05-08:
@M95D can you help me out with how you built it (I'm 2.6.36brcm / armv7)? My jffs2 partition is still messed up unfortunately.

M95D, 2020-05-09:
Sure.
[mtd-utils-2.1.1-brcmarm.tar.gz](https://github.com/Entware/Entware/files/4604896/mtd-utils-2.1.1-brcmarm.tar.gz)

KyleSanderson, 2020-05-14:
Thanks fellow human, I owe you a beer.

closed 2020-09-02 by ryzhovau