Entware/Entware #443: msmtpq requires bash

davygravy, 2020-03-30:
I installed msmtpq/msmtp-queue and it works, but the basic executable script located at /opt/bin/msptpq seems to have a typo in it.  

Line 1 reads:
`#! /bin/sh`
which of course causes the script to be executed with the built-in Busybox shell, which is `ash`.

Invoking it gives:
`
root@host:/tmp/home/root# msmtpq
/opt/bin/msmtpq: line 117: declare: not found
/opt/bin/msmtpq: line 118: declare: not found
/opt/bin/msmtpq: line 357: syntax error: bad substitution`

The `declare` function is not included in `ash`, but is included in `bash`.

Changing line 1 to 
`#! /opt/bin/bash` fixed it for me.

I am -not- sure that this is the best way to fix it.

ryzhovau, 2020-03-30:
Thanks for report. We should find some graceful way to solve it.

davygravy, 2020-03-30:
Thanks Alexander!  Cheers & stay well!

ryzhovau, 2020-03-31:
The easiest way is to install bash and change shebang strings.
The best way is to rewrite those script with no bash'ism.

The upstream already [tried](https://github.com/openwrt/packages/issues/11547#issuecomment-597427384) to get best way with no success, so we pick the easiest way for now.

eigood, 2020-07-07:
#!/usr/bin/env bash

Then normal $PATH lookup occurs, and finds bash in /opt.

ryzhovau, 2020-08-30:
Since 1.8.7 `msmtp-queue` requires bash.

closed 2020-08-30 by ryzhovau