Entware/Entware #444: Radvd isn't working

Lichtenshtein, 2020-04-01:
```
Apr 1 10:53:19 radvd[4006]: version 2.18 started
Apr 1 10:53:19 radvd[4006]: exiting, permissions on conf_file invalid
Apr 1 11:25:59 radvd[6733]: version 2.17 started
Apr 1 11:25:59 radvd[6733]: exiting, permissions on conf_file invalid
```

Changed permissions in different ways - all the same.

The package http://bin.entware.net/mipselsf-k3.4/radvd_2.18-1_mipsel-3.4.ipk

ryzhovau, 2020-04-01:
Please, [take a look how permissions checked](https://github.com/reubenhwk/radvd/blob/849f69c3e6876fad009a8f3be27d7533ce9a56db/radvd.c#L859). 

* if username isn't set from command line, it defaults to root,
* `/opt/etc/radvd.conf` must not be writable by self/own group.

Just checked on Padavan's firmware where 0 user isn't a `root`, works fine.
```
# touch /opt/etc/radvd.conf
# chmod 440 /opt/etc/radvd.conf
# radvd --nodaemon --username=admin
[Apr 01 18:06:43] radvd (11478): version 2.18 started
/opt/etc/radvd.conf:1 error: syntax error, unexpected $end, expecting T_INTERFACE
[Apr 01 18:06:43] radvd (11478): exiting, failed to read config file
```

closed 2020-04-01 by ryzhovau

Lichtenshtein, 2020-04-02:
> Please, [take a look how permissions checked](https://github.com/reubenhwk/radvd/blob/849f69c3e6876fad009a8f3be27d7533ce9a56db/radvd.c#L859).

I googled it but did not understand correctly.

>  --username=admin

Thanks for that, now it starts.