Entware/Entware #447: OpenSSL issue

przemasik, 2020-04-08:
Until now, I used Let's Encrypt to generate a certificate from this script https://github.com/Entware/Entware-ng/wiki/Using-Let's-Encrypt. Unfortunately, for several weeks he has been throwing the following errors.
I do not know what's going on.

`przemas@RT-AC86U:/tmp/mnt/opt/entware/etc/nginx# bash ./dehydrated --domain server.xxx.com --cron
# INFO: Using main config file /opt/etc/nginx/config
547645430992:error:25066067:DSO support routines:dlfcn_load:could not load the shared library:crypto/dso/dso_dlfcn.c:118:filename(libengines.so): libengines.so: cannot open shared object file: No such file or directory
547645430992:error:25070067:DSO support routines:DSO_load:could not load the shared library:crypto/dso/dso_lib.c:162:
547645430992:error:0E07506E:configuration file routines:module_load_dso:error loading dso:crypto/conf/conf_mod.c:224:module=engines, path=engines
547645430992:error:0E076071:configuration file routines:module_run:unknown module name:crypto/conf/conf_mod.c:165:module=engines
548063608016:error:25066067:DSO support routines:dlfcn_load:could not load the shared library:crypto/dso/dso_dlfcn.c:118:filename(libengines.so): libengines.so: cannot open shared object file: No such file or directory
548063608016:error:25070067:DSO support routines:DSO_load:could not load the shared library:crypto/dso/dso_lib.c:162:
548063608016:error:0E07506E:configuration file routines:module_load_dso:error loading dso:crypto/conf/conf_mod.c:224:module=engines, path=engines
548063608016:error:0E076071:configuration file routines:module_run:unknown module name:crypto/conf/conf_mod.c:165:module=engines
548319153360:error:25066067:DSO support routines:dlfcn_load:could not load the shared library:crypto/dso/dso_dlfcn.c:118:filename(libengines.so): libengines.so: cannot open shared object file: No such file or directory
548319153360:error:25070067:DSO support routines:DSO_load:could not load the shared library:crypto/dso/dso_lib.c:162:
548319153360:error:0E07506E:configuration file routines:module_load_dso:error loading dso:crypto/conf/conf_mod.c:224:module=engines, path=engines
548319153360:error:0E076071:configuration file routines:module_run:unknown module name:crypto/conf/conf_mod.c:165:module=engines
548384595152:error:25066067:DSO support routines:dlfcn_load:could not load the shared library:crypto/dso/dso_dlfcn.c:118:filename(libengines.so): libengines.so: cannot open shared object file: No such file or directory
548384595152:error:25070067:DSO support routines:DSO_load:could not load the shared library:crypto/dso/dso_lib.c:162:
548384595152:error:0E07506E:configuration file routines:module_load_dso:error loading dso:crypto/conf/conf_mod.c:224:module=engines, path=engines
548384595152:error:0E076071:configuration file routines:module_run:unknown module name:crypto/conf/conf_mod.c:165:module=engines
Processing server.xxx.com
 + Signing domains...
 + Generating private key...
 + Generating signing request...
Error configuring OpenSSL modules
548134366416:error:25066067:DSO support routines:dlfcn_load:could not load the shared library:crypto/dso/dso_dlfcn.c:118:filename(libengines.so): libengines.so: cannot open shared object file: No such file or directory
548134366416:error:25070067:DSO support routines:DSO_load:could not load the shared library:crypto/dso/dso_lib.c:162:
548134366416:error:0E07506E:configuration file routines:module_load_dso:error loading dso:crypto/conf/conf_mod.c:224:module=engines, path=engines
548134366416:error:0E076071:configuration file routines:module_run:unknown module name:crypto/conf/conf_mod.c:165:module=engines
`przemas@RT-AC86U:/tmp/mnt/opt/entware/etc/nginx#``

ryzhovau, 2020-04-08:
Seems like openssl misconfiguration. Try to comment out this string at `/opt/etc/ssl/openssl.cnf`:
```
[openssl_conf]
#engines=engines
```

przemasik, 2020-04-09:
> Seems like openssl misconfiguration. Try to comment out this string at `/opt/etc/ssl/openssl.cnf`:
> 
> ```
> [openssl_conf]
> #engines=engines
> ```

Very thanks! It's works!
I'm asking for closing the thread.

closed 2020-04-09 by przemasik