Entware/Entware #451: Installing libjpeg (9c-2) does not Install

pavian57, 2020-04-16:
Bug Report
---

Collected errors:
 * opkg_install_pkg: Package size mismatch: libjpeg is 114776 bytes, expecting 114798 bytes
 * opkg_install_cmd: Cannot install package libjpeg.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

ryzhovau, 2020-04-16:
Which feed do you have (mipsel, armv5 and so on)?
Did you `opkg update` before install?

pavian57, 2020-04-16:
First answer was wrong, from this place

https://bin.entware.net/x64-k3.2/

Regards

ryzhovau, 2020-04-17:
Can't reproduce it on fresh install. Please, run `opkg update first`.
```
$ opkg install libjpeg
Installing libjpeg (9c-2) to root...
Downloading http://bin.entware.net/x64-k3.2/libjpeg_9c-2_x64-3.2.ipk
Configuring libjpeg.
```

closed 2020-04-17 by ryzhovau