Entware/Entware #453: Wireguard-go

dulemis, 2020-04-17:
Hi!

I have successfully installed wireguard-go_nohf on my router. The interface seems to star tup, but opkg install wireguard-tools is missing. However, on the website package list I can find wireguard-tools.

So i've tried manually downloading the ipkg-file from the entware website

opkg install /jffs/wireguard-tools_1.0.20200206-1_armv7-3.2.ipk
Unknown package 'wireguard-tools'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for wireguard-tools found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package wireguard-tools.

zyxmon, 2020-04-17:
missing wireguard-tools package is now in repo. Please install it with `opkg install wireguard-tools`.
fixed via https://github.com/Entware/Entware/commit/276badd2f3c420251d303482f6e86af1b34f0505

closed 2020-04-17 by zyxmon

dulemis, 2020-04-18:
I don't know if this maybe is the wrong place, but I can install both packages, configure it and run it. But after transferring data over the tunnel, wireguard-go crashes with the message "segmentation fault".
Any ideas why this is?

zyxmon, 2020-04-18:
Try do add swap space. We know wireguard-go works for mispel devices with a new kernel.

dulemis, 2020-04-18:
I have an ARM-router.
I have enabled a swap file onthe same USB-devicem where entware is

Total / Free Memory	249.63 MB / 205.47 MB (82.31%)
Total / Free Swap	256.00 MB / 256.00

It doesnt seem to solve the problem. Any other ideas, please?

zyxmon, 2020-04-18:
You can try previous version of packages from archive subfolder.
Probably this is a folder for your hardware - http://bin.entware.net/armv7sf-k2.6/archive/

dulemis, 2020-04-18:
There is no older version than 0.0.20200121...

0x5e, 2020-05-18:
Same issue. When I'm transferring data, It crashed. I'm using `wireguard-go_nohf - v0.0.20200121-1` installed on ASUS AC68U with merlin system.
Here's my log:
```
xxx@RT-AC68U-5DB0:/jffs/entware# wireguard-go -f wg0
WARNING WARNING WARNING WARNING WARNING WARNING WARNING
W                                                     G
W   You are running this software on a Linux kernel,  G
W   which is probably unnecessary and misguided. This G
W   is because the Linux kernel has built-in first    G
W   class support for WireGuard, and this support is  G
W   much more refined than this slower userspace      G
W   implementation. For more information on           G
W   installing the kernel module, please visit:       G
W           https://www.wireguard.com/install         G
W                                                     G
WARNING WARNING WARNING WARNING WARNING WARNING WARNING
INFO: (wg0) 2020/05/18 15:50:19 Starting wireguard-go version 0.0.20200121
INFO: (wg0) 2020/05/18 15:50:19 Interface set up
INFO: (wg0) 2020/05/18 15:50:19 Device started
INFO: (wg0) 2020/05/18 15:50:19 UAPI listener started
Segmentation fault

xxx@RT-AC68U-5DB0:/jffs/entware# cat /jffs/entware/etc/entware_release 
release=entware
arch=arm
cpu=cortex-a9
cpu_subtype=unknown
float=soft
gcc=8.3.0
gcc_flags=-O2 -pipe -mtune=cortex-a9 -fno-caller-saves -mfloat-abi=soft
libc=glibc
libc_version=2.23

xxx@RT-AC68U-5DB0:/jffs/entware# uname -a
Linux RT-AC68U-5DB0 2.6.36.4brcmarm #1 SMP PREEMPT Tue Sep 25 18:58:07 CST 2018 armv7l Merlin-Koolshare

xxx@RT-AC68U-5DB0:/jffs/entware# opkg list-installed | grep wireguard
wireguard-go_nohf - v0.0.20200121-1
wireguard-tools - 1.0.20200206-1
```

zyxmon, 2020-05-18:
It looks like wireguard-go is incompatible with old (2.x) kernels. We will remove this packages from the repo in case there are no success stories.

0x5e, 2020-05-18:
Update:
I build wireguard-go using: `GOARM=5 GOARCH=arm GOOS=linux CGO_ENABLED=0 go build -ldflags "-extldflags '-static' -w -s " -o wireguard-go-armv5` and the problem solved.

[wireguard-go-armv5.zip](https://github.com/Entware/Entware/files/4645490/wireguard-go-armv5.zip)

https://linkscue.com/posts/2019-11-14-go-illegal-instruction-error/

zyxmon, 2020-05-18:
wireguard-go_nohf is the version for broadcom CPU with no hardfloat support. It is build with GOARM=5 flag! Use this version for Asus routers.

https://github.com/Entware/entware-go/blob/master/wireguard-go/Makefile#L28

dulemis, 2020-05-18:
> Update:
> I build wireguard-go using: `GOARM=5 GOARCH=arm GOOS=linux CGO_ENABLED=0 go build -ldflags "-extldflags '-static' -w -s " -o wireguard-go-armv5` and the problem solved. [...]

Thank you! 
This one seem to work without segmentstion fault, however it's slower then Strongswan IPsec :(

0x5e, 2020-05-19:
> wireguard-go_nohf is the version for broadcom CPU with no hardfloat support. It is build with GOARM=5 flag! Use this version for Asus routers.
> 
> https://github.com/Entware/entware-go/blob/master/wireguard-go/Makefile#L28

This version for me is illegal instruction :)
And I found it's slow too :)

0x5e, 2020-05-19:
[wireguard-go-armv5~v7.zip](https://github.com/Entware/Entware/files/4647831/wireguard-go-armv5.v7.zip)
Here's my build for test, if anyone need.

```
GOARM=5 GOARCH=arm GOOS=linux CGO_ENABLED=0 go build -ldflags "-extldflags '-static' -w -s " -o wireguard-go-armv5
GOARM=6 GOARCH=arm GOOS=linux CGO_ENABLED=0 go build -ldflags "-extldflags '-static' -w -s " -o wireguard-go-armv6
GOARM=7 GOARCH=arm GOOS=linux CGO_ENABLED=0 go build -ldflags "-extldflags '-static' -w -s " -o wireguard-go-armv7
```

ASUS AC68U can only run wireguard-go-armv5 version.

zyxmon, 2020-05-19:
@0x5e - can you make PR for our makefile?

zyxmon, 2020-05-19:
@0x5e - which go compiler are yuo using?
I have updated package wireguard-go to 1a in the repo. I have used compilation flags from **0x5e** - they make binary smaller but are in fact nearly the same as mine (there is a small difference).

Plz check the new version in the repo. If it still crashes I can upgrade go compiler from 1.14 to 1.14.3 and compile wireguard-go again.

0x5e, 2020-05-19:
I'm using `go version go1.13.11 linux/arm64` in another linux router(nanopi r2s). Not sure what I do is correct.
Although wireguard-go-armv5 works for me, but it's still not usable. It will exit with no reason.

```
xxx@RT-AC68U-5DB0:/tmp/home/root# cat /tmp/wireguard-go.log
INFO: (wg0) 2020/05/19 09:44:42 Starting wireguard-go version 0.0.20200320-22-gf60b391
INFO: (wg0) 2020/05/19 09:44:42 Device started
INFO: (wg0) 2020/05/19 09:44:42 UAPI listener started
INFO: (wg0) 2020/05/19 09:44:42 Interface set up
INFO: (wg0) 2020/05/19 09:50:27 Device closing
INFO: (wg0) 2020/05/19 09:50:27 Interface closed
INFO: (wg0) 2020/05/19 09:50:27 Shutting down
```