Entware/Entware #454: Ingenic T10

daniel-dona, 2020-04-17:
Python, Python3 and Node.js installs ok but can only get a segfault at runtime.

¿Maybe a different MIPS(EL) ABI?

```
[root@DafangHacks:~]# node
Segmentation fault
[root@DafangHacks:~]# python
Segmentation fault
[root@DafangHacks:~]# python3
Segmentation fault
[root@DafangHacks:~]# 
```

```
[33146.167811] do_page_fault() #2: sending SIGSEGV to node for invalid read access from
[33146.167811] 77762624 (epc == 776348f8, ra == 776141d8)
[33149.003350] do_page_fault() #2: sending SIGSEGV to python for invalid read access from
[33149.003350] 7769f624 (epc == 7755e8f8, ra == 7753e1d8)
[33151.300541] do_page_fault() #2: sending SIGSEGV to python3 for invalid read access from
[33151.300541] 76deb624 (epc == 76caa8f8, ra == 76c8a1d8)
```

```
[root@DafangHacks:~]# cat /proc/cpuinfo 
system type		: mango
machine			: Unknown
processor		: 0
cpu model		: Ingenic Xburst V0.0  FPU V0.0
BogoMIPS		: 858.52
wait instruction	: yes
microsecond timers	: no
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 1, address/irw mask: [0x0fff]
isa			: mips32r1
ASEs implemented	:
shadow register sets	: 1
kscratch registers	: 7
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

Hardware		: isvp
Serial			: 00000000 00000000 00000000 00000000
```

zyxmon, 2020-04-17:
`Maybe a different MIPS(EL) ABI?` - opkg would not work if your architecture is not correct. Probably your kernel is too old.

daniel-dona, 2020-04-17:
Kernel:

`Linux DafangHacks 3.10.14 #92 PREEMPT Mon Aug 6 00:05:24 CEST 2018 mips GNU/Linux`

Also, ¿what are the MIPSEL ABI target?

zyxmon, 2020-04-17:
Compilation flags
MIPS (BE) - https://github.com/Entware/Entware/blob/master/configs/mips-3.4.config#L189
MIPS (LE) - https://github.com/Entware/Entware/blob/master/configs/mipsel-3.4.config#L188

Please reinstall Entware and show your installation log on pastebin.

ryzhovau, 2020-04-17:
Looks like it's MIPS32 Release 1 ISA CPU. 
We compile MIPS binaries with MIPS R2 instruction set.

daniel-dona, 2020-04-17:
> Looks like it's MIPS32 Release 1 ISA CPU.
> We compile MIPS binaries with MIPS R2 instruction set, which is incompatible with R1.
> 
> All we can offer is our deprecated MIPS R1 feed http://old.entware.net/binaries/entware/

Oh, I see, so... in theory, I could try to compile the current packages changing the flag to mips32r1, Am I right with that?

ryzhovau, 2020-04-18:
That's right. 
Try those old feed first, just to be sure.

closed 2020-04-18 by ryzhovau

daniel-dona, 2020-04-18:
Feeds didn't work, I don't know why, but I was compiling the packages by myself at the same time so I didn't care so much.
I can confirm that packages install and work fine now changing the compilation flag to "-mips32 -mtune=mips32", so I thing is solved.

Thank you so much everyone!