Entware/Entware #455: SoftEther VPN server v5.01.9673 security fix release available.

Myron-S, 2020-04-18:
SoftEther VPN server v5.01.9673 security fix release.
---
{Fix security issue: Fix the security of JSON-API. If the administrator password of the Virtual Hub is empty}

- [] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [] I'm ready to test added package on my {platform} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

ryzhovau, 2020-04-18:
Duplicate of #456

closed 2020-04-18 by ryzhovau

Myron-S, 2020-08-12:
What is published on Entware is V5 SoftEther and it is this that this speciffic request was opened to ask for an update.  I have asked upstream for the appropriate packege to be made available for `Linux NAS 3.10.38 #1 SMP Thu Aug 29 20:49:31 PDT 2019 Build-gitcdc15cb x86_64 GNU/Linux` here [https://github.com/openwrt/packages/issues/13095](https://github.com/openwrt/packages/issues/13095).