Entware/Entware #456: Make stable version of SoftEther VPN server available on Entware.

Myron-S, 2020-04-18:
SoftEther VPN (Stable Edition Repository): Open Cross-platform Multi-protocol VPN Software. This branch is officially managed by Daiyuu Nobori, the owner of SoftEther VPN Project.
---
{The stable edition of SoftEther VPN is not available on Entware. Can this be included so stable or developer version can be selected for install?}

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my {platform} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

ryzhovau, 2020-04-18:
It's up to [upstream](https://github.com/openwrt/packages/tree/master/net/softethervpn). 
Please, [ask](https://github.com/openwrt/packages/issues) OpenWrt guys to update it.

closed 2020-04-18 by ryzhovau

Myron-S, 2020-08-31:
I'm told from the OpenWRT side; "Entware problem. Version 9674 is already available here."
Who on entware need to get this and do what is needed to get this on the entware repo?

ryzhovau, 2020-08-31:
We'll update SoftEther VPN in next decade or so.