Entware/Entware #457: ooRexx and Fail2Ban

tkalfaoglu, 2020-04-19:
I would love to have these two packages..
ooRexx is great for system scripts
and Fail2ban is great against hackers..

ryzhovau, 2020-04-19:
What a killer feature oRexx have against other script languages? 

We didn't pack `fail2ban` into repo as long as it's too heavy for most of embedded devices. You can always install necessary dependencies via pip.

tkalfaoglu, 2020-04-20:
It's just that I have written programs in the tens of thousands of lines in Rexx so I  am comfortable with it. I will try to compile it on my AC68U.. 
Thank you I found the fail2ban.. Indeed it is CPU and I/O intensive..

tkalfaoglu, 2020-04-21:
Hi - I compiled ooRexx 5.0B and it works well on my AC68U..  But I am not sure how to package it for inclusion into the repository so others can use it...

ryzhovau, 2020-06-23:
Closing for now.

closed 2020-06-23 by ryzhovau