Entware/Entware #459: ASUS GT-AX11000 missing /opt/var link

aselvan, 2020-04-19:
Entware install problem on ASUS GT-AX11000 stock firmware
---
The stock firmware is missing the readonly link (/opt/var) so Entware install fails with the following messages. Is there any way to work around this? The ASUS stock firmware I am on is:  Version:3.0.0.4.384_8011

```
Info: Basic packages installation...
Collected errors:
 * make_directory: Cannot create directory `/opt/var': Read-only file system.
 * make_directory: Cannot create directory `/opt/var': Read-only file system.
 * make_directory: Cannot create directory `/opt/var': Read-only file system.
```
```
/tmp/home/root# opkg update
Collected errors:
 * make_directory: Cannot create directory `/opt/var': Read-only file system.
 * make_directory: Cannot create directory `/opt/var': Read-only file system.
 * make_directory: Cannot create directory `/opt/var': Read-only file system.

```
I attempted to edit the opkg.conf to work around it but still not able to get past the missing  /opt/var 

```
/tmp/home/root# opkg -f /opt/etc/opkg.conf update
Downloading http://bin.entware.net/aarch64-k3.10/Packages.gz
Updated list of available packages in /opt/tmp/opkg/opkg-lists/entware
/tmp/home/root# opkg -f /opt/etc/opkg.conf install tcpdump
Installing tcpdump (4.9.3-1) to root...
Collected errors:
 * verify_pkg_installable: Only have 0kb available on filesystem /opt, pkg tcpdump needs 363
 * opkg_install_cmd: Cannot install package tcpdump.
```

ryzhovau, 2020-04-20:
Please, ask other ASUS users on [SNB forum](https://www.snbforums.com/forums/asus-wireless.37/). It's up to firmware.

closed 2020-04-20 by ryzhovau

aselvan, 2020-04-20:
> Please, ask other ASUS users on [SNB forum](https://www.snbforums.com/forums/asus-wireless.37/). It's up to firmware.

Thanks, I will post it in that forum. I totally understand the problem is with ASUS firmware issue. I was just wondering if anyone here found any work around.

themiron, 2020-04-21:
as workaround you can try to remount the whole /opt to rw partion or to some r/w dir with -o bind

aselvan, 2020-04-21:
Great tip. I got the /opt mounted directly under the /sda1 and its all **good**. 

`/dev/sda1 on /opt type ext3 (rw,relatime,data=ordered)
`

Just need to add the mount line to the post usb mount script to nvram. I am good to go. 

```
> nvram set script_usbmount="/jffs/post-mount"
> nvram commit
```

Thank you for the suggestion! I knew someone would have a workaround here.

chinaa108, 2020-08-15:
Hi aselvan, I’ve tried to follow your guide on your page but I still am not able to install entware. I keep getting /opt/bin no such file or directory. Etc. Please help

aselvan, 2020-08-15:
> Hi aselvan, I’ve tried to follow your guide on your page but I still am not able to install entware. I keep getting /opt/bin no such file or directory. Etc. Please help

As indicated above, this is not an Entware issue and is already closed. If you need help, I have my script to mount USB drive with the workaround above posted in Github at the link below which you are welcome to copy and customise it. Once you mount your USB with this script, Entware should install just fine. If it doesn't work, you can enter a [comment here](https://github.com/aselvan/scripts/issues) w/ your complete log file, error message etc and I can then help you over there.

[https://github.com/aselvan/scripts/blob/master/router/asus-gt-ax11000/asus_usbmount.sh](url)