Entware/Entware #463: How to build a mirror site for bin.entware.net?

JACK-THINK, 2020-04-29:
Hello,

How to build a mirror site for bin.entware.net? Is there any script or other method to sync?

Thanks a lot.

alexcom, 2020-05-02:
I suppose info in [this issue](https://github.com/Entware/Entware-ng/issues/292) still applies. Correct me if I’m wrong but then the only way to get yourself a mirror is to build all packages yourself.

ryzhovau, 2020-05-02:
Chinese guys eventually blocked by Cloudflare, I've [noticed recently](https://t.me/entware/2707). 

1) there's HTTPS only access to `bin.entware.net`,
2) avoid `bin.entware.net/<arch>/archive` mirroring, that's just an old packages even been present in repo,
3) please, apply some "soft" limits on your side, such as speed/threads of download and/or update periods.

JACK-THINK, 2020-05-02:
@alexcom and @ryzhovau 
Thanks for your reply!

@ryzhovau 
Will the script below work without consuming too much resource of your server?

```shell
#!/bin/sh
#
# Crontab configuration
# 0 3 * * 3 "absolute_path_to_this_script"
#
ENTWARE_REPO="/var/www/entware_repo"
#
if [ ! -d "${ENTWARE_REPO}" ]
then
    mkdir -p "${ENTWARE_REPO}"
fi
#
cd "${ENTWARE_REPO}"
#
for ARCH in "aarch64-k3.10" "armv5sf-k3.2" "armv7sf-k2.6" "armv7sf-k3.2" "mipselsf-k3.4" "mipssf-k3.4" "other" "x64-k3.2" "x86-k2.6"
do
    {
        rm -rf "${ENTWARE_REPO}/${ARCH}"
        #
        wget --no-check-certificate --tries=10 --timeout=100 --limit-rate=5m --continue --no-host-directories --recursive --exclude-directories="/${ARCH}/archive/" --no-parent "https://bin.entware.net/${ARCH}/"
        #
    } &
done
#
wait
```

ryzhovau, 2020-05-02:
1. Try to not run it in parallel `... } &`
2. Don't forget `/css/packages.css` used for `Packages.html`
3. Better to find a way to update existing mirror instead of download content every time.

closed 2020-05-02 by ryzhovau

JACK-THINK, 2020-05-04:
@ryzhovau 

Will the script below work without consuming too much resource of your server?

```shell

#!/bin/sh
#
# Crontab configuration
# 0 3 * * 3 "<absolute_path_to_this_script>"
#
########## PUBLIC VARIABLES ##########
#
ENTWARE_REPO_HOST="https://bin.entware.net"
ENTWARE_REPO_ROOT="<path to local entware repo>"
#
########## END ##########
#
#
########## LOAD FUNCTIONS ##########
#
createDirectoryTree()
{
    local TARGET_DIRECTORY="${1}"
    local TEMP_DIRECTORY="${ENTWARE_REPO_ROOT}/sync_tmp"
    local PRESENT_DIRECTORY_TREE="${TEMP_DIRECTORY}/present_directory_tree"
    local LATEST_DIRECTORY_TREE="${TEMP_DIRECTORY}/latest_directory_tree"
    #
    rm -rf "${TEMP_DIRECTORY}"
    #
    mkdir -p "${TEMP_DIRECTORY}"
    #
    wget --timestamping --no-check-certificate --tries=10 --timeout=100 --limit-rate=5m --continue --no-host-directories --recursive --no-parent --accept "index.html" --directory-prefix="${TEMP_DIRECTORY}" "${ENTWARE_REPO_HOST}"
    #
    find "${TEMP_DIRECTORY}" -type d | sed -E -e '\#/archive/?[^/]*#d' -e 's#'"${TEMP_DIRECTORY}"'#'"${ENTWARE_REPO_ROOT}"'#' -e 's#/$##' > "${LATEST_DIRECTORY_TREE}"
    find "${ENTWARE_REPO_ROOT}" -type d | sed -E -e '\#'"${TEMP_DIRECTORY}"'#d' -e 's#/$##' > "${PRESENT_DIRECTORY_TREE}"
    #
    grep -Fxvf "${PRESENT_DIRECTORY_TREE}" "${LATEST_DIRECTORY_TREE}" | grep -E -e '^'"${TARGET_DIRECTORY}"'' > "${TEMP_DIRECTORY}/list_to_mkdir"
    grep -Fxvf "${LATEST_DIRECTORY_TREE}" "${PRESENT_DIRECTORY_TREE}" | grep -E -e '^'"${TARGET_DIRECTORY}"'' > "${TEMP_DIRECTORY}/list_to_remove"
    #
    for ITEM in $(cat "${TEMP_DIRECTORY}/list_to_mkdir")
    do
        mkdir -p "${ITEM}" 2> /dev/null
    done
    #
    for ITEM in $(cat "${TEMP_DIRECTORY}/list_to_remove")
    do
        rm -rf "${ITEM}"
    done
    #
    rm -rf "${TEMP_DIRECTORY}"
}
#
fileSync()
{
    local LOCAL_DIRECTORY="${1}"
    local TEMP_DIRECTORY="${LOCAL_DIRECTORY}/sync_tmp"
    local URL_DIRECTORY="$(echo "${LOCAL_DIRECTORY}" | sed -e 's#'"${ENTWARE_REPO_ROOT}"'/*##')"
    local PRESENT_INDEX="${LOCAL_DIRECTORY}/index.html"
    local CURRENT_INDEX="${TEMP_DIRECTORY}/current_index.html"
    local LATEST_INDEX="${TEMP_DIRECTORY}/latest_index.html"
    #
    rm -rf "${TEMP_DIRECTORY}"
    #
    mkdir -p "${TEMP_DIRECTORY}"
    #
    wget --no-check-certificate --tries=10 --timeout=100 --limit-rate=5m --continue --no-host-directories --output-document="${LATEST_INDEX}" "${ENTWARE_REPO_HOST}/${URL_DIRECTORY}/"
    #
    if [ -e "${PRESENT_INDEX}" ]
    then
        cp -f "${PRESENT_INDEX}" "${CURRENT_INDEX}"
    else
        echo "" > "${CURRENT_INDEX}"
    fi
    #
    grep -Fxvf "${CURRENT_INDEX}" "${LATEST_INDEX}" | sed -E -n -e 's#^<a href=\"([^/]+)\".*#\1#p' > "${TEMP_DIRECTORY}/list_to_download"
    grep -Fxvf "${LATEST_INDEX}" "${CURRENT_INDEX}" | sed -E -n -e 's#^<a href=\"([^/]+)\".*#\1#p' > "${TEMP_DIRECTORY}/list_to_remove"
    #
    for ITEM in $(cat "${TEMP_DIRECTORY}/list_to_download")
    do
        wget --timestamping --no-check-certificate --tries=10 --timeout=100 --limit-rate=5m --continue --directory-prefix="${LOCAL_DIRECTORY}" "${ENTWARE_REPO_HOST}/${URL_DIRECTORY}/${ITEM}"
    done
    #
    for ITEM in $(cat "${TEMP_DIRECTORY}/list_to_remove")
    do
        rm -f "${LOCAL_DIRECTORY}/${ITEM}"
    done
    #
    cp -f "${LATEST_INDEX}" "${PRESENT_INDEX}"
    #
    rm -rf "${TEMP_DIRECTORY}"
}
#
########## END ##########
#
#
########## MAIN ##########
#
DIRECTORY_TO_SYNC="${ENTWARE_REPO_ROOT}/<specific architecture or blank>"
#
mkdir -p "${ENTWARE_REPO_ROOT}/css/" 2> /dev/null
wget --no-check-certificate --tries=10 --timeout=100 --limit-rate=5m --continue --no-host-directories --output-document="${ENTWARE_REPO_ROOT}/css/packages.css" "${ENTWARE_REPO_HOST}/css/packages.css"
#
createDirectoryTree "${DIRECTORY_TO_SYNC}"
#
DIRECTORY_LIST="$(find "${DIRECTORY_TO_SYNC}" -type d | sed -e 's#/$##')"
#
for ITEM in ${DIRECTORY_LIST}
do
    fileSync "${ITEM}"
done
#
########## END ##########

```

wlnirvana, 2020-08-16:
@JACK-THINK Any update about your mirror?