Entware/Entware #465: About CUPS  requesting your  Help,

kisbyee, 2020-05-04:
My device is a router. The model is YOUKU YK1. It supported miniSD,so I installed one for it.I've been trying to turn my router into a print server.The router firmware I use is "openwrt-19.07.2-ramips-mt7620-youku-yk1-squashfs-sysupgrade.bin".then，Then I set up the Entware environment on the SD card，and，opkg update && opkg  install cups ghostscript cups-filter (cups - 2.2.12-1).
I forgot to say，My printer is a USB printer，The model is Lenovo LJ2400.This printer works properly on Ubuntu 18.04 CUPS,Even though it uses “ BROTHER HL-2170W FOOMATIC/HL1250”, It works very good.
When I have done the preliminary and necessary setting,I can open localhost: 631 on my browser。but, Administrarion->Add printer->Local Printers is nothing.  I try a lot，but it didn't get better。I think it's a problem。
Yesterday, when I didn't install the entware environment，I uploaded cups2.1.4 and its dependencies through winscp. And installed them. Administrarion->Add printer->Local Printers ,that looks normal.I really hope he can work with ghostscript，But I didn't find the right ghostscript package.
When I send the document of "application / PDF" through IPP protocol. He will report a mistake ：'client-error-document-format-not-supported‘ ，and so on.I've seen https://github.com/Entware/Entware/issues/270,But I can't solve it.

ryzhovau, 2020-05-04:
We aren't ready to provide personal support, sorry.

closed 2020-05-04 by ryzhovau

whc2001, 2024-05-13:
Same here, nothing is discovered when adding printer through WebUI and this appears in the error log:

```
...
I [13/May/2024:19:44:17 +0000] [Client 1] Started "/opt/lib/cups/cgi-bin/admin.cgi" (pid=3847, file=12)
I [13/May/2024:19:44:17 +0000] [Client 2] Started "/opt/lib/cups/daemon/cups-deviced" (pid=3848, file=14)
E [13/May/2024:19:44:18 +0000] [CGI] ippfind (PID 3863) stopped with status 2!
E [13/May/2024:19:44:18 +0000] [cups-deviced] PID 3855 (driverless) stopped with status 2!
```

Seems like CUPS in Entware is broken. You need to use this: https://github.com/TheMMcOfficial/lede-cups.git