Entware/Entware #469: SSHUTTLE

Stryp, 2020-05-14:
Package request
---
Hello,

I really would like to have SSHUTTLE package 
https://github.com/sshuttle/sshuttle

This is really useful when you want  to forward all traffic (or specific subnets) from my local network (served by the router) to a remote server transparently without configure proxy on the clients. 

- [ x] No similar package already added and/or this one got some new killer feature(s),
- [ x] Package is in active development (at least, not abandoned for 5 years or more),
- [ x] I'm ready to test added package on my {platform} and get a feedback about it.

---

zyxmon, 2020-05-15:
This is a python package - try to install it yourself - https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules

Stryp, 2020-05-15:
Hi zyxmo,

Thank you for your answer, you're right indeed I managed to install it, but seems that I need also the following two packages which are not in the repo: 

 * opkg_install_cmd: Cannot install package iptables-mod-nat-extra.
 * opkg_install_cmd: Cannot install package iptables-mod-ipopt.

Stryp, 2020-05-15:
I just tried to start SSHUTTLE and it works for what I need!

closed 2020-05-16 by zyxmon