Entware/Entware #473: Pls update mini-snmpd to 1.6

KrellPatch, 2020-05-31:
Mini-snmpd shows some major feature additions since 1.5. and is currently at 1.6.  Entware currently provides 1.4. An update would be most welcome. 

Cheers, and stay safe

arigit, 2020-08-19:
Huge +1 on this :)
v1.4 only supports 32bit counters and it has some strange issues with them, occasionally misbehaving on counter rollover.
v1.6 supports 64 bit counters and does away with those issues.

I use entware on asus merlin / wrt router

ryzhovau, 2020-09-02:
It's up to [upstream](https://github.com/openwrt/packages/blob/master/net/mini_snmpd/Makefile).
Please, ask OpenWrt guys to update it and newer version will arrive here.

closed 2020-09-02 by ryzhovau

arigit, 2020-09-02:
@ryzhovau thanks - upstream issue opened

arigit, 2020-09-26:
@ryzhovau upstream was updated a week ago to 1.6, trying opkg update on my entware install still doesn't show the mini-snmpd version. Wondering if entware needs to manually pull the updated version from upstream? or does entware periodically resync with upstream