Entware/Entware #474: pgsql-server and python

cat4er, 2020-05-31:
I try to create web server on my Xiaomi R3P based on Django, Postgres, Nginx & Gunicorn.
One of depends package is psycopg2 (adapter for python and postgres), but I can't install it via pip without programm pg_config.

Third party forum has information about package libpq-dev, which decide my problem, I didn't find in entware repo.

are there any plans to create it? can u help me?

> 
> `MI-R3P:/opt/home/admin # pip install psycopg2
> Collecting psycopg2
>   Using cached psycopg2-2.8.5.tar.gz (380 kB)
>     ERROR: Command errored out with exit status 1:
>      command: /opt/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/opt/tmp/pip-install-s0yx2__q/psycopg2/setup.py'"'"'; __file__='"'"'/opt/tmp/pip-install-s0yx2__q/psycopg2/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /opt/tmp/pip-pip-egg-info-bul9wp84
>          cwd: /opt/tmp/pip-install-s0yx2__q/psycopg2/
>     Complete output (23 lines):
>     running egg_info
>     creating /opt/tmp/pip-pip-egg-info-bul9wp84/psycopg2.egg-info
>     writing /opt/tmp/pip-pip-egg-info-bul9wp84/psycopg2.egg-info/PKG-INFO
>     writing dependency_links to /opt/tmp/pip-pip-egg-info-bul9wp84/psycopg2.egg-info/dependency_links.txt
>     writing top-level names to /opt/tmp/pip-pip-egg-info-bul9wp84/psycopg2.egg-info/top_level.txt
>     writing manifest file '/opt/tmp/pip-pip-egg-info-bul9wp84/psycopg2.egg-info/SOURCES.txt'
>     
>     Error: pg_config executable not found.
>     
>     pg_config is required to build psycopg2 from source.  Please add the directory
>     containing pg_config to the $PATH or specify the full executable path with the
>     option:
>     
>         python setup.py build_ext --pg-config /path/to/pg_config build ...
>     
>     or with the pg_config option in 'setup.cfg'.
>     
>     If you prefer to avoid building psycopg2 from source, please install the PyPI
>     'psycopg2-binary' package instead.
>     
>     For further information please check the 'doc/src/install.rst' file (also at
>     <https://www.psycopg.org/docs/install.html>).
>     
>     ----------------------------------------
> ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
> `

cat4er, 2020-06-15:
https://github.com/Entware/entware-packages/pull/12

ryzhovau, 2020-08-30:
As long as we use OpenWrt buildroot, there's no any `*-dev` packages because it's useless on target systems.

You may try to transfer `pg_config` binary from buildroot to target system and [use native compilation](https://github.com/Entware/Entware/wiki/Using-GCC-for-native-compilation) which is quite painful.

PS We do not provide personal support, sorry.

closed 2020-08-30 by ryzhovau