Entware/Entware #480: Working ftptop ("no curses or ncurses library on this system")

virtualdj, 2020-06-07:
I'm running Entware on a QNAP x86_64 NAS and I would like to run **ftptop** which is part of **proftpd - 1.3.7rc3-1 package**.
However, it doesn't work:
```
[~] # which ftptop
/opt/bin/ftptop
[~] # ldd /opt/bin/ftptop
        linux-vdso.so.1 (0x00007ffcb93d2000)
        libattr.so.1 => /opt/lib/libattr.so.1 (0x00007ff834366000)
        libcrypt.so.1 => /opt/lib/libcrypt.so.1 (0x00007ff83432c000)
        libgcc_s.so.1 => /opt/lib/libgcc_s.so.1 (0x00007ff834313000)
        libc.so.6 => /opt/lib/libc.so.6 (0x00007ff834160000)
        /opt/lib/ld-linux-x86-64.so.2 (0x00007ff83436f000)
[~] # ftptop
ftptop: no curses or ncurses library on this system
[~] # opkg list_installed | grep curses
libncurses - 6.2-1
libncursesw - 6.1-5
python3-ncurses - 3.7.4-2
```
Is there any other dependency required? I cannot guess why it fails... does it depend on the kernel itself?

zyxmon, 2020-06-09:
fixed via https://github.com/Entware/entware-oldpackages-ports/commit/70c3592b4f76ca06507d4f910ecf92138eed6610
Binaries will be uploaded with upcoming update

closed 2020-06-09 by zyxmon

virtualdj, 2020-06-27:
Unfortunately, it still doesn't work... 
```
[~] # opkg list_installed | grep proftp
proftpd - 1.3.7rc3-2
[~] # opkg list_installed | grep curses
libncurses - 6.2-1
libncursesw - 6.2-1
python3-ncurses - 3.8.3-2
[~] # which ftptop
/opt/bin/ftptop
[~] # ldd /opt/bin/ftptop
        linux-vdso.so.1 (0x00007ffc7b981000)
        libncursesw.so.6 => /opt/lib/libncursesw.so.6 (0x00007fef1c07f000)
        libattr.so.1 => /opt/lib/libattr.so.1 (0x00007fef1c078000)
        libcrypt.so.1 => /opt/lib/libcrypt.so.1 (0x00007fef1c03e000)
        libgcc_s.so.1 => /opt/lib/libgcc_s.so.1 (0x00007fef1c024000)
        libc.so.6 => /opt/lib/libc.so.6 (0x00007fef1be70000)
        /opt/lib/ld-linux-x86-64.so.2 (0x00007fef1c0f2000)
[~] # ftptop
ftptop: no curses or ncurses library on this system
```

zyxmon, 2020-06-28:
Cannot confirm this behaviour. Try to
```opkg update
opkg upgrade
opkg remove --autoremove proftpd
opkg install proftpd
```

virtualdj, 2020-06-28:
You're right... I do not know why, but yesterday it gave me that error (in fact I copied and pasted from the terminal), while now without doing anything (neither reinstalling) it works... Sorry!
```
[~] # ftptop
ftptop/0.9: Sun Jun 28 12:05:12 2020, up for 14 hrs 56 min
0 Total FTP Sessions: 0 downloading, 0 uploading, 0 idle

PID   S USER     CLIENT               SERVER          TIME COMMAND
```
Thank you very much!