Entware/Entware #483: Avahi daemon fails on dd-wrt

silverman10, 2020-06-12:
Installed entware from 
wget http://bin.entware.net/mipssf-k3.4/installer/generic.sh for dd-wrt as per the linked wiki.

opkg update and opkg install avahi-utils.

Upon start of avahi daemon (rc.unslung start), avahi-daemon fails.

`Jun 12 10:04:26 DD-WRT user.notice root: Started dbus-daemon from.
Jun 12 10:04:26 DD-WRT user.notice root: Started avahi-daemon from.
Jun 12 10:04:26 DD-WRT daemon.err avahi-daemon[2643]: mkdir("/run/avahi-daemon/"): No such file or directory `

`root@DD-WRT:/opt# etc/init.d/rc.unslung check                                                                            Checking dbus-daemon...              alive.                                                                             
Checking avahi-daemon...              dead.`

```
root@DD-WRT:/opt# opkg list avahi-utils                                                                                 
avahi-utils - 0.8-1 - Avahi is an mDNS/DNS-SD (aka RendezVous/Bonjour/ZeroConf) 
```

cvonk, 2020-06-15:
Same here.  /run is read-only

zyxmon, 2020-06-16:
Please make sure you run `/opt/sbin/avahi-daemon` - this binary should not use /run directory.

scharan, 2020-06-16:
Seems like it does:
```
root@dd-wrt:~# /opt/sbin/avahi-daemon
mkdir("/run/avahi-daemon/"): No such file or directory
root@dd-wrt:~# opkg list avahi-utils
avahi-utils - 0.8-1 - Avahi is an mDNS/DNS-SD (aka RendezVous/Bonjour/ZeroConf)
root@dd-wrt:~# uname -a
Linux saitm 4.4.220 #2 SMP Wed Apr 29 02:56:57 +04 2020 armv7l DD-WRT
```

zyxmon, 2020-06-16:
Please create /opt/var/run folder in case id does not exist and try again.

scharan, 2020-06-16:
That exists too:
```
root@dd-wrt:~# ls -l /opt/var/
total 4
drwxr-xr-x    2 root     root          1024 Mar 23 00:23 lock
drwxr-xr-x    2 root     root          1024 Mar 23 00:23 log
drwxr-xr-x    2 root     root          1024 Jun 15 21:49 opkg-lists
drwxr-xr-x    3 root     root          1024 Jun 15 22:30 run
```

cvonk, 2020-06-17:
dbus seems to be using /opt/var/run

root@rtr:~# strings `which avahi-daemon` | grep run/
/run/avahi-daemon//pid
/run/avahi-daemon/
mkdir("/run/avahi-daemon/"): %s
Failed to create runtime directory /run/avahi-daemon/.
/run/avahi-daemon/socket

zyxmon, 2020-06-17:
We will upload entware repo update in few days with the fix.

mrachell, 2020-06-17:
unsubscribe

cvonk, 2020-06-17:
@zyxmon, much appreciated.

scharan, 2020-06-17:
Thank you, @zyxmon.

cvonk, 2020-06-22:
Thanks, the update did the trick.  

```
root@rtr:/jffs# avahi-daemon
Found user 'nobody' (UID 99) and group 'nogroup' (GID 114).
Successfully dropped root privileges.
avahi-daemon 0.6.31 starting up.
WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
dbus_bus_get_private(): Failed to connect to socket /opt/var/run/dbus/system_bus_socket: No such file or directory
WARNING: Failed to contact D-Bus daemon.
avahi-daemon 0.6.31 exiting.
root@rtr:/jffs# vi /opt/etc/avahi/
avahi-daemon.conf  services/
root@rtr:/jffs# vi /opt/etc/avahi/avahi-daemon.conf
```

Had to set `enable-dbus=no` in `/opt/etc/avahi/avahi-daemon.conf`

```
root@rtr:/jffs# avahi-daemon
Found user 'nobody' (UID 99) and group 'nogroup' (GID 114).
Successfully dropped root privileges.
avahi-daemon 0.6.31 starting up.
WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
No service file found in /opt/etc/avahi/services.
socket() failed: Address family not supported by protocol
Failed to create IPv6 socket, proceeding in IPv4 only mode
Joining mDNS multicast group on interface br1.IPv4 with address 10.1.1.1.
New relevant interface br1.IPv4 for mDNS.
Joining mDNS multicast group on interface br0.IPv4 with address 10.0.1.1.
New relevant interface br0.IPv4 for mDNS.
Joining mDNS multicast group on interface vlan2.IPv4 with address xxxxxxxx.
New relevant interface vlan2.IPv4 for mDNS.
Network interface enumeration completed.
Registering new address record for 10.1.1.1 on br1.IPv4.
Registering new address record for 10.0.1.1 on br0.IPv4.
Registering new address record for 169.254.255.1 on br0.IPv4.
Registering new address record for xxxxxxxxx on vlan2.IPv4.
Registering HINFO record with values 'ARMV7L'/'LINUX'.
Server startup complete. Host name is rtr.local. Local service cookie is xxxxxxx.
```

Voila: success!

scharan, 2020-06-23:
Thanks, works for me too. I also needed @cvonk's finding, I had to set `enable-dbus=0` in `avahi-daemon.conf` under the `[server]` section.

ryzhovau, 2020-06-23:
Thanks for feedback.

closed 2020-06-23 by ryzhovau