Entware/Entware #485: Python 2.7.x missing?

OneCDOnly, 2020-06-18:
Hey @zyxmon. :)

QNAP NAS TS-559 Pro+ running QTS 4.2.6

I've just started getting errors when attempting to install the older Python 2.7.x via Entware.

Here's a from-scratch install: 
```
[/share/Public] # wget http://bin.entware.net/other/Entware_1.02std.qpkg
--2020-06-19 08:56:07--  http://bin.entware.net/other/Entware_1.02std.qpkg
Resolving bin.entware.net (bin.entware.net)... 104.27.176.50, 104.27.177.50, 172.67.212.134, ...
Connecting to bin.entware.net (bin.entware.net)|104.27.176.50|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 27382 (27K) [application/x-gzip]
Saving to: ‘Entware_1.02std.qpkg’

Entware_1.02std.qpkg                    100%[==============================================================================>]  26.74K  91.1KB/s   in 0.3s   

2020-06-19 08:56:08 (91.1 KB/s) - ‘Entware_1.02std.qpkg’ saved [27382/27382]
```

```
[/share/Public] # sh Entware_1.02std.qpkg 
Install QNAP package on TS-NAS...
7+1 records in
7+1 records out
./
./built_info
./qinstall.sh
./package_routines
./qpkg.cfg
0+1 records in
0+1 records out
3+1 records in
3+1 records out
Firmware check is fine.
Sym-link /opt ...
Info: Opkg package manager deployment...
Info: Basic packages installation...
Downloading http://bin.entware.net/x64-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
Installing entware-opt (227000-3) to root...
Downloading http://bin.entware.net/x64-k3.2/entware-opt_227000-3_all.ipk
Installing libgcc (8.4.0-9) to root...
Downloading http://bin.entware.net/x64-k3.2/libgcc_8.4.0-9_x64-3.2.ipk
Installing libc (2.27-9) to root...
Downloading http://bin.entware.net/x64-k3.2/libc_2.27-9_x64-3.2.ipk
Installing libssp (8.4.0-9) to root...
Downloading http://bin.entware.net/x64-k3.2/libssp_8.4.0-9_x64-3.2.ipk
Installing libpthread (2.27-9) to root...
Downloading http://bin.entware.net/x64-k3.2/libpthread_2.27-9_x64-3.2.ipk
Installing librt (2.27-9) to root...
Downloading http://bin.entware.net/x64-k3.2/librt_2.27-9_x64-3.2.ipk
Installing libstdcpp (8.4.0-9) to root...
Downloading http://bin.entware.net/x64-k3.2/libstdcpp_8.4.0-9_x64-3.2.ipk
Installing entware-release (1.0-2) to root...
Downloading http://bin.entware.net/x64-k3.2/entware-release_1.0-2_all.ipk
Installing zoneinfo-asia (2020a-1) to root...
Downloading http://bin.entware.net/x64-k3.2/zoneinfo-asia_2020a-1_x64-3.2.ipk
Installing zoneinfo-europe (2020a-1) to root...
Downloading http://bin.entware.net/x64-k3.2/zoneinfo-europe_2020a-1_x64-3.2.ipk
Installing findutils (4.7.0-1) to root...
Downloading http://bin.entware.net/x64-k3.2/findutils_4.7.0-1_x64-3.2.ipk
Installing terminfo (6.2-1) to root...
Downloading http://bin.entware.net/x64-k3.2/terminfo_6.2-1_x64-3.2.ipk
Installing libpcre (8.43-2) to root...
Downloading http://bin.entware.net/x64-k3.2/libpcre_8.43-2_x64-3.2.ipk
Installing grep (3.4-1) to root...
Downloading http://bin.entware.net/x64-k3.2/grep_3.4-1_x64-3.2.ipk
Installing locales (2.27-9) to root...
Downloading http://bin.entware.net/x64-k3.2/locales_2.27-9_x64-3.2.ipk
Installing opkg (2020-05-07-f2166a89-1) to root...
Downloading http://bin.entware.net/x64-k3.2/opkg_2020-05-07-f2166a89-1_x64-3.2.ipk
Installing entware-upgrade (1.0-1) to root...
Downloading http://bin.entware.net/x64-k3.2/entware-upgrade_1.0-1_all.ipk
Configuring libgcc.
Configuring libc.
Configuring libssp.
Configuring libpthread.
Configuring librt.
Configuring terminfo.
Configuring libpcre.
Configuring grep.
Configuring locales.
Entware uses separate locale-archive file independent from main system
Creating locale archive /opt/usr/lib/locale/locale-archive
Adding en_EN.UTF-8
Adding ru_RU.UTF-8
You can download locale sources from http://bin.entware.net/other/i18n_glib227.tar.gz
You can add new locales to Entware using /opt/bin/localedef.new
Configuring entware-upgrade.
Upgrade operations are not required.
Configuring opkg.
Configuring zoneinfo-europe.
Configuring zoneinfo-asia.
Configuring libstdcpp.
Configuring entware-release.
Configuring findutils.
Configuring entware-opt.
Link service start/stop script: Entware.sh
Set QPKG information in /etc/config/qpkg.conf
[App Center] Installed Entware-std 1.02 in /share/MD0_DATA/.qpkg/Entware.
Enable Entware/opkg
[App Center] Enabled Entware-std.
```
Then logout SSH and back in.
```
[~] # opkg install python
Unknown package 'python'.
Collected errors:
 * opkg_install_cmd: Cannot install package python.
```
None of the Python 2.7.x packages are shown in `opkg list`.

Help!

zyxmon, 2020-06-19:
Repository was updated on the 18th of June. Python2 was removed from the repo. This was Openwrt decision to remove Python2.

OneCDOnly, 2020-06-19:
Oh damn... that's going to cause a few problem. :(

closed 2020-06-19 by OneCDOnly