Entware/Entware #486: Urgent: bin.entware.net down

milankragujevic, 2020-06-19:
[image]

You can see it's not a problem on my end because this is a Cloudflare 522 (Gateway Timeout) error. Same from multiple IPs. 2020-06-19 17:25:00 UTC+2

[image]

Regular site entware.net (on GH Pages I presume) works. 

This is a huge problem because my device is in the middle of installing Entware when this died, and if I turn it off or there is a power outage it will be bricked.

ryzhovau, 2020-06-19:
Thanks for report! 

Our hosting is down since 2020-06-19 15:25:00 UTC after hardware failure. I hope we'll resolve it soon, stay tuned.

milankragujevic, 2020-06-19:
Can you send me a zip of the "http://bin.entware.net/armv7sf-k2.6/installer" folder, I'll host it locally. I am risking my router's "life", I do not want it to break, I can't reflash it with UART right now. Thanks!

milankragujevic, 2020-06-19:
@zyxmon Actually no, I am using a custom one, it literally only needs these files:

```
http://bin.entware.net/armv7sf-k2.6/installer/opkg
http://bin.entware.net/armv7sf-k2.6/installer/opkg.conf
http://bin.entware.net/armv7sf-k2.6/installer/ld-2.23.so
http://bin.entware.net/armv7sf-k2.6/installer/libc-2.23.so
http://bin.entware.net/armv7sf-k2.6/installer/libgcc_s.so.1
http://bin.entware.net/armv7sf-k2.6/installer/libpthread-2.23.so
```

zyxmon, 2020-06-19:
If fact now you need only opkg and opkg.conf 
They are here = http://sw.zyxmon.org/tmp/
Installation script is here - https://github.com/Entware/installer.sh

PS You need to edit opkg.conf and generic.sh to point to correct URL

zyxmon, 2020-06-19:
opkg bynary is static now and does not need *.so files that are left for old scripts compatibility.

milankragujevic, 2020-06-19:
Thank you so much. For others, if your links are gone and this happens again:

[opkg.zip](https://github.com/Entware/Entware/files/4805793/opkg.zip)

xRose1, 2020-06-19:
I thought it was me that was having issues. Thanks for the updates!

TCB13, 2020-06-19:
+1 on this.

````
wget http://bin.entware.net/armv7sf-k3.2/installer/generic.sh
Connecting to bin.entware.net (172.67.212.134:80)
wget: server returned error: HTTP/1.1 522 Origin Connection Time-out
````

2b, 2020-06-20:
So, there was only one server, without any mirror, right?

Ubimo, 2020-06-20:
bin.entware.net is up again!

zyxmon, 2020-06-20:
Fixed.

closed 2020-06-20 by zyxmon

fahaad9, 2020-07-01:
Still the same. i tried running opkg update but get wget returned 8.