Entware/Entware #487: Error when issuing opkg upgrade command

slyx101, 2020-06-20:
Running on Asustor AS5304T with Entware installed from the Asustor AppCentral.

It was working find until a recent package update (opkg update).
The update itself worked fine (below):
  Downloading https://bin.entware.net/x64-k3.2/Packages.gz
  Updated list of available packages in /opt/var/opkg-lists/entware

However, when attempting an update of the package, the following errors were encountered:
 * opkg_download: Failed to download https://bin.entware.net/x64-k3.2/x64-k3.2/libssp_8.4.0-9_x64-3.2.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download libssp. Perhaps you need to run 'opkg update'?
 * opkg_download: Failed to download https://bin.entware.net/x64-k3.2/x64-k3.2/git-http_2.26.2-1_x64-3.2.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download git-http. Perhaps you need to run 'opkg update'?
 * opkg_download: Failed to download https://bin.entware.net/x64-k3.2/x64-k3.2/libopenssl_1.1.1g-1_x64-3.2.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download libopenssl. Perhaps you need to run 'opkg update'?

Checking the URLs above, seems that opkg is attempting to download the packages from the wrong location (i.e. https://bin.entware.net/x64-k3.2/x64-k3.2/) What is the cause of the double entry of "x64-k3.2"?

Is this due to any recent change in the packages file which could have caused this? Hope someone can help.

Much thanks
Slyx

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

Avian-Flew, 2020-06-20:
Experiencing exactly the same issue here.

okpg.conf file states
src/gz entware http://bin.entware.net/x64-k3.2
dest root /
dest ram /opt/tmp
lists_dir ext /opt/var/opkg-lists
option tmp_dir /opt/tmp
arch all 100
arch x64-3.x 150
arch x64-3.2 160

zyxmon, 2020-06-20:
Now it is fixed.
The issue is caused by a hardware failure in ВС and switching to the mirror.
https://github.com/Entware/Entware/issues/486

closed 2020-06-20 by zyxmon