Entware/Entware #491: package archive missing?

OneCDOnly, 2020-06-21:
Hi guys. :nerd_face: 

It looks like the `archive` path is now missing. Is this a permanent change? 

Example: `http://bin.entware.net/x64-k3.2/archive` is not there anymore. 

Any chance of making these available again? I have a current package that only runs on Python 3.7.4 :disappointed: so I need to force-downgrade by using files from this archive.

Thank you.

ryzhovau, 2020-06-21:
Only main feeds of `bin.entware.net` is back online. Archive, sources and other stuff will be added later.

OneCDOnly, 2020-06-21:
Thanks @ryzhovau . 

I hate to ask, but can you suggest when that will be? So I can keep people up-to-date.

Cheers.

ryzhovau, 2020-06-21:
ETA is June, 23, but can't promise that, sorry.

OneCDOnly, 2020-06-21:
Oh, that's not so bad.

 Cheers mate! :smile:

xRose1, 2020-06-21:
thanks too.

ryzhovau, 2020-06-22:
Archive is back online.

closed 2020-06-22 by ryzhovau