Entware/Entware #493: Could you, please, include the transmission with the sequential support to entware?

lokapal, 2020-06-22:
Hello, developers/maintainers!

      Could you, please, include (or replace) the "Transmission" package to the version with the sequential support? This feature lowers CPU load many times and is very important at relatively weak CPUs (that is the main entware target CPUs). The transmission fork with this feature is available [just here ](https://github.com/Mikayex/transmission)

I'm ready to test added/modified package on my armv7sf-k3.2 platform and get a feedback about it.

konst2000, 2020-06-22:
This will be extremely useful. many people need it

ryzhovau, 2020-06-23:
Please, use `transmission-cfp`, sequental download feature is present there.

I see no sense to add third variant of transmission, sorry.

closed 2020-06-23 by ryzhovau