Entware/Entware #496: sysstat: /opt/lib/sysstat/sa2 is not compatible with Asuswrt-Merlin

ColinTaylorUK, 2020-06-24:
/opt/lib/sysstat/sa2 will crash and therefore not create the daily reports.

/opt/lib/sysstat/sa2 requires the following packages to be installed and the script modified accordingly:

findutils: to support find's "-type f"
coreutils-date: to support date's "--date=yesterday"
xz: Not present in Asuswrt-Merlin but script can be changed to use gzip.

[diff.txt](https://github.com/Entware/Entware/files/4826116/diff.txt)

Side note: /opt/etc/sysstat/sysstat is not included in the package, but it's not actually needed.

ryzhovau, 2020-08-30:
AFAIK `$PATH` on asuswrt-merlin contains `/usr/bin` before `/opt/bin` for some reason, to preserve some firmware specific behavior. 

Also, that for forgotten `/opt/etc/sysstat/sysstat`, I'll send a PR for upstream.

closed 2020-08-30 by ryzhovau

ColinTaylorUK, 2020-08-30:
Thanks you.

Sorry if this is a stupid question but in your commit you have added a dependency for `coreutils-find`. My Entware installation doesn't have this package it has `findutils` instead. 

 https://github.com/Entware/entware-packages/blob/e11d4e9ab25878dcfa482a1f8a12df1a7ca67920/utils/sysstat/Makefile#L34

ryzhovau, 2020-08-30:
My bad. Thanks again.